goodtft/LCD-show #101: chmod -R 755 LCD-show

wimismith, 2018-10-20:
i was setting my pi up following the directions, and after i got through the second step, it couldn't find chmod -R 755 LCD-show. i looked into the file labeled lcd-show and its empty. what can i do to resolve this issue?

lavolp3, 2018-10-22:
The LCD-show that is referred to here is a folder, not a file. You clone
the project in the first step and the folder LCD-show is created in your
working directory. Then in the second step you chmod the LCD-show folder to
have the proper permissions.

ForgivenNin, 2019-02-01:
When I tried running that command, for some reason my Pi didn't say there was the proper permissions to run the command (probably as you said @lavolp3 ). How exactly do you do what you said?

If it helps any for me to include this - I was using Raspbian Stretch Lite