goodtft/LCD-show #105: OMXPlayer doesn't display output via LCD-Show

ambanmba, 2018-12-07:
Have the LCD working fine with LCD-Show and things seem to work properly except when I run OMXPlayer the videos do not appear on the LCD screen and instead are sent out to HDMI only. Is this an expected behaviour?

ambanmba, 2018-12-14:
Per this link, `omxplayer --display 4` should do the trick, but it does not.
https://www.raspberrypi.org/documentation/raspbian/applications/omxplayer.md

Anonymousdog, 2018-12-18:
> Per this link, `omxplayer --display 4` should do the trick, but it does not.
> https://www.raspberrypi.org/documentation/raspbian/applications/omxplayer.md

[](url)

> Per this link, `omxplayer --display 4` should do the trick, but it does not.
> https://www.raspberrypi.org/documentation/raspbian/applications/omxplayer.md

[https://www.raspberrypi.org/forums/viewtopic.php?t=199525#p1244506](https://www.raspberrypi.org/forums/viewtopic.php?t=199525#p1244506) Suggests that omxplayer plain doesn't support these screens.

...but you might be able to use this tool to work around the issue, [https://www.raspberrypi.org/forums/viewtopic.php?t=63133](https://www.raspberrypi.org/forums/viewtopic.php?t=63133)