goodtft/LCD-show #106: 3.5‘’ incorrect touch

JesonWang09, 2018-12-09:
my 3.5'' screen on Raspbian official mirror, but it got incorrect display.
The X-axis and the Y-axis have opposite information.
such as, when i touch right side of the screen, but the cursor appears at the top of the screen.
What can I do to solve this problem?

Jason0416, 2018-12-12:
使用Raspberrypi Host name安装

JesonWang09, 2018-12-13:
> 使用Raspberrypi Host name安装

我用的默认的pi用户通过sudo 提权安装的，和你说的安装方式有什么区别吗

saholman, 2018-12-27:
I'm having this same issue. Is this a hardware issue or a software issue?

saholman, 2019-01-08:
This solved my issue: https://github.com/swkim01/waveshare-dtoverlays#touch-calibration

I simply added Option "TransformationMatrix" "0 -1 1 1 0 0 0 0 1" to /etc/X11/xorg.conf.d/99-calibration.conf to rotate touch 90 degrees to the left.

rimeno, 2019-02-14:
Same problem, solved by @saholman comment. Thanks

JesonWang09, 2019-03-04:
yeah， @saholman ‘s solution is effective. tks!

closed 2019-03-04 by JesonWang09