goodtft/LCD-show #107: LCD 3.5" screen keep showing boot log

Alexxxxissssssss, 2018-12-10:
Hi, 
i have an issue with the 3.5" screen, i've installed everything correctly on the last version of raspbian, after doing ./LCD35-show the board reboot, the console boot is running on the screen but when everything is done the log stays on the screen. But when i click on something, it appears that the desktop is behind and i can see it briefly before the log comes back...

juancarmo, 2018-12-30:
I have this problem too, dont know how to solve this issue.

aha-swipe, 2019-01-15:
Try removing the following line from **/etc/rc.local**

`
fbcp &
`

What is this doing? When using a 3.5" TFT the raspberry creates two framebuffers (memory where the picture that is displayed on a screen is stored), one for the HDMI output and the other for the TFT. fbcp is a tool that constantly copies the data from the HDMI buffer to the TFT buffer to increase display performance. In my case this was constantly overwriting the content for the TFT with the console and not what I wanted. Removing those lines fixed the issue.

RottenPi, 2019-02-17:
> Try removing the following line from **/etc/rc.local**
>
> `fbcp &`
>
> What is this doing? When using a 3.5" TFT the raspberry creates two framebuffers (memory where the picture that is displayed on a [...]

I have this problem too, but there is no `fbcp &` in the file you said....

aha-swipe, 2019-02-18:
Then you probably have a different problem, what does your rc.local file say?

sceic, 2019-06-26:
I have this problem too

goodtft, 2019-06-29:
try install again?
2019-06-29 

goodtft

marianorenzi, 2019-08-12:
> Try removing the following line from **/etc/rc.local**
>
> `fbcp &`
>
> What is this doing? When using a 3.5" TFT the raspberry creates two framebuffers (memory where the picture that is displayed on a [...]

This solved the issue for me. Thanks!

andrewboni, 2019-09-09:
Commenting out `fbcp &` from `/etc/rc.local` also resolved this for me!

davelnewton, 2019-09-13:
Commenting out `fbcp` also (effectively) disables `_rotate...` directives in `config.txt`. 

You'll need to update the `dtoverlay` to reflect desired rotation if the default isn't what you need.

KEINOS, 2019-11-17:
Another vote for **commenting out `fbcp &` from `/etc/rc.local` 👍 This solved my issue** ! Thanks!

- H/W: BCM2835, Rev: 9000c1, Model: **Raspberry Pi Zero W** Rev 1.1
- OS: Raspbian **Buster Lite** ([2019-07-10-raspbian-buster-lite](https://downloads.raspberrypi.org/raspbian_lite/images/raspbian_lite-2019-07-12/).zip), No X-Window installed
- Display: MHS35, 3.5'' LCD Display, TS11-JP
- Driver: Installed. (`clone` the repo -> `cd LCD-show` -> `sudo ./MHS35-show`. Though it gave me errors of `xserver-xorg-input-evdev` but after the reboot LCD started to display the boot log) 
- Issue: The boot log on the LCD get stuck as below:
  ```
  ...
  [6.464968] systemd[1]: Created slice system-systemd\x2dfsck.slice.
  [6.518587] systemd[1]: Created slice syste-serial\x2dgetty.slice.
  [6.568936] systemd[1]: Created slice system-getty.slice.
  ```
- Solution: Even the console log on LCD gets stuck, it is accessible to RPi-Zero W via SSH. So I **commented out as [suggested](https://github.com/goodtft/LCD-show/issues/107#issuecomment-454471063) and rebooted**, then it fixed my problem. TTY1 works as usual.
- Note: I'm not using GUI (Raspbian Desktop) but CUI only (Raspbian Lite with no X). So don't know if the touch sensor work.

mahad-ahmed, 2024-04-26:
> Try removing the following line from **/etc/rc.local**
>
> `fbcp &`
>
> What is this doing? When using a 3.5" TFT the raspberry creates two framebuffers (memory where the picture that is displayed on a [...]

This did it for me! I was using the script from the similar Waveshare repository BTW. The screen kept blanking out and only showed when there was keyboard input.