goodtft/LCD-show #109: Is there a way to get HDMI and LCD to work?

ralfstoermer, 2018-12-24:
Is there a way to get HDMI and LCD-Tochscreen to work simultan?
Does the screen support fbtft?

Thanks!

AndyGaskell, 2019-03-18:
Any ideas on this?  It would be great to have screen mirroring.

BrayanKhosravian, 2019-04-07:
at least it should be able to deactivate the tft mode

goodtft, 2019-07-01:
@ralfstoermer I don't understand what you said.Isn't HDMI and LCD-Touchscreen not working synchronously?

BrayanKhosravian, 2019-07-01:
It does not work synchronously for me

SilasMueller, 2019-11-27:
Ive got the same question. Is there no solution to this?

webash, 2020-04-17:
I'm assuming we're unlikely to get a response here, but did anyone already in this thread get it working?

webash, 2020-04-27:
I did; if anyone is interested, message me

SilasMueller, 2020-04-28:
> 
> 
> I did; if anyone is interested, message me

could you post the solution here? could help others in the future too

webash, 2020-05-02:
No worries, I wrote this to help:

https://ashley.geek.nz/2020/05/02/dual-screen-on-a-raspberry-pi-with-hdmi-goodtft-gpio-touchscreens/

AythaE, 2021-05-26:
@webash Your blog seems to be down, it gives me a `Error establishing a database connection`. Can you share this dual screen tutorial on a different way

webash, 2021-06-30:
@AythaE apologies for the delay, the site should be online again now :)

elektronika-ba, 2023-06-03:
The blog is down again (page not found), but a copy on Archive.org is available:

https://web.archive.org/web/20211020140334/https://ashley.geek.nz/2020/05/02/dual-screen-on-a-raspberry-pi-with-hdmi-goodtft-gpio-touchscreens/

webash, 2023-08-08:
Yeah sorry about that. Need to shift my blog off WordPress but y'all know how personal tech debt can be... 😅