goodtft/LCD-show #112: [Rpi 3B+ - Sticky Finger Kali] White screen

ReWeda, 2019-01-03:
Hi!
I've got a Raspberry pi 3B+ with Sticky Finger (Kali Linux w/ Re4son Kernel) installed. 
I'm trying to correctly install a 3.5'' Raspberry LCD.

Once connected my Raspberry to HDMI screen and booting up I followed the instruction: fetching from git, changing permission and running the executable file (LCD35-show in my case).
Running the executable file resulted in a couple of command that I can't even read 'cause the Raspberry suddenly reboot (the output was something like 'unable ...more stuff')
Once rebooted, using an hdmi screen no GUI loads, just cmd line; connecting the LCD screen instead results in persistent white screen.

I reinstalled Sticky finger and tried to run LCD35-show with the LCDscreen attached. Same results.

I already checked [#98](https://github.com/goodtft/LCD-show/issues/98#issue-367420018) and [#61](https://github.com/goodtft/LCD-show/pull/61#issuecomment-427661572).
I still have to try #61 proposed solution. Do I have to try that way w/ or w/o the LCDscreen connected?

Any other advice?

chmtt, 2019-04-01:
Same issue here..

goodtft, 2019-07-01:
see this:
https://github.com/lcdwiki/LCD-show-kali