goodtft/LCD-show #115: Kali no GIU

hadri1983, 2019-01-22:
I'm having the exact same issues as #48 https://github.com/goodtft/LCD-show/issues/48

Tried different versions of Kali including the official and Sticky Finger. As soon as the good TFT driver is installed, there is no was to get the GUI back.

Good TFT mentioned there was a Kali version with the driver pre-installed, can someone please direct me to it.

Ev1lpete, 2019-01-24:
I took the Stickyfingers base image, the goodTFT 3.5 inch touchscreen, selected the Waveshare screens from the base drivers (3.5A) - haven't tried B yet - But it works. Rotate the screen either 90 or 270 and the GUI login fires up. (HDMI output showed the initial bootup until gui launched.

MrWolf-to, 2019-01-29:
Hi, I've just tried with latest Kali Linux and made it work after some adjustment. I have a Kuman 3.5'' LCD GPIO and tried on Raspberry Pi 3 Model B+.
If I've understood your issue, to have the GUI back, try this:

sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf

and enter these strings:

Section "Device"
        Identifier "myfb"
        Driver "fbdev"
        Option "fbdev" "/dev/fb1"
EndSection

goodtft, 2019-07-01:
see this:
https://github.com/lcdwiki/LCD-show-kali

bmxbiker206, 2022-01-25:
Hello from the future..

TL;DR version: PiTFT/Pi4 Kali ARM GUI shows momentarily then black screen after LCD-Show and adding info to 99-fbdev.conf. Parts of GUI show when touched before disappearing again. 

Fresh Kali-Linux-ARM install on Pi4. Connect to WiFi using monitor connected to HDMI.

```
sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
```

then

`sudo apt-get install tightvncserver -y`

SSH and VNC servers working over WiFi. Great!

Has 3.5" TFT. Maybe Waveshare.. maybe not. I didn't purchase it and I really can't tell if it is a knock off or not. 

Try `sudo kalipi-tft-config`

Every.. possible.. option.. update overlays.. Nothing good coming. Ah but maybe LCD-Show?

```
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show
```

And the screen comes to life.. but where is that beautiful GUI? The above suggestion is the only thing that made any difference at all but here is the rub.. First the fix as stated by user @MrWolf-to :

> sudo nano /usr/share/X11/xorg.conf.d/99-fbdev.conf
> 
> and enter these strings:
> 
> Section "Device"
> Identifier "myfb"
> Driver "fbdev"
> Option "fbdev" "/dev/fb1"
> EndSection

There is the GUI but only for a brief moment. Its there.. I can tell it is. If I scroll around the touchscreen I can see parts of it appear before disappearing once again. As it sits I only see a black screen with single static dash in the top left corner. 

Beside 99-fbdev I see in kalip-tft-config these when running show configuration

```
/boot/cmdline.txt:
dwc_otg.fiq_fix_enable=2 console=ttyAMA0,115200 kgdboc=ttyAMA0,115200 console=tty1 root=PARTUUID=88db9d9b-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait net.ifnames=0

/etc/X11/xorg.conf.d/99-calibration.conf:
Section "InputClass"
Identifier      "calibration"
MatchProduct    "ADS7846 Touchscreen"
Option  "Calibration"   "227 3936 3880 268"
Option  "SwapAxes"      "1"
EndSection
```

I can also tell the touch calibration is of by 90 degree but I can fix that once I can see the GUI.

bmxbiker206, 2022-02-04:
@MrWolf-to @goodtft @Ev1lpete @hadri1983 Sadly my post is not getting any attention. Sorry for tagging you all. I was just hoping someone had an idea on how to fix. Thinking I am going to try pasting the original config and see what happens.

MrWolf-to, 2022-02-04:
Sorry, I don't have a Raspberry Pi 4. :/ I stopped to Raspberry Pi 3 Model B+.