goodtft/LCD-show #116: Kuman 3.5'' LCD GPIO doesn't work with Raspbian on Raspberry Pi W Zero

MrWolf-to, 2019-01-29:
Hi!
I bought a Kuman 3.5'' LCD (connected to GPIO) and it works great with latest Raspbian on Raspberry Pi Model B+ (and, with some adjustament, with Kali Linux), but I cannot make it work with Raspberry Pi W Zero: I get a white screen.
Anybody experienced the same issue?

WesBrenner, 2019-03-03:
Hi:
I also have trouble with the latest raspbian version (stretch) and tracked it down to an attempt to use some of the multi-core routines which would not be used by the zero.  It can't find them because the zero uses the single core kernel and it would not load the multi-core routines.

MrWolf-to, 2019-03-03:
Hi, actually, the problem was that the GPIO header was not soldered, and keeping it "floating" doesn't make the screen work.
I bought a Pi Zero W with header soldered and the screen works without issue (both Raspbian and Kali).