goodtft/LCD-show #122: No Signal after running LCD5-show

gsansone4, 2019-02-28:
I cloned the repo and ran chmod 755 -R LCD-show, ran sudo ./LCD5-show, but haven't been able to get any signal from my board. Board is a Firefly ROC-RK3328-CC (64-bit ARM, very similar to RPi3) running the ubuntu 16.04 image from the firefly website. I tried the 18.04 image as well with no luck. When I plug an HDMI into my Windows laptop, the LCD displays fine but with no touch (not an issue, I just want the display working).

Thanks in advance

Allexin, 2019-03-05:
Same thing with RPi3 and LCD7C

gsansone4, 2019-03-05:
Not exactly a fix but it worked with the Android image.

ladrians, 2019-03-30:
I have the same issue using xenial Ubuntu 16.04.6 LTS, with LCD7B-show.
Any detail on how to troubleshoot it?

Chip-Tronics, 2019-04-09:
I have the same exact issue.
Raspberry pi 3B, Retropie 4.4.11.
On the last line of code to input, cp: cannot create blah blah blah.
Then the Rpi3 reboots and screen is still "No Signal".
I've contacted the seller "Juvtmall, he's of no use. He offered no explanation whatsoever.
I'll never do business with them again. I should have gotten a definite list of possible fixes or reasons. He offered nothing.
There's one part of the installation that doesn't jive with the sale page instructions.
There's one point where the system asks for a reboot for the driver to take effect. I don't know if I reboot or not and keep forging ahead with the installation. I tried both ways to no avail.
Then at the last line of code, I hit enter and the system shuts down, no driver install, no picture on the screen.
The CD documentation gives 2 ways to install.
I even tried what they have in documentation of installing the on CD Raspian, which they CLAIM already has the drivers installed.
NOPE!
No picture on the display.
Any help on this would be very much appreciated!

robdejonge, 2019-05-28:
@gsansone4 See https://github.com/goodtft/LCD-show/issues/140, the `LCD5-show` script is horrendous. You're better off picking it apart and executing the commands yourself, which it sounds like you would be knowledgeable enough to do. Hope you've resolved your issue by now though!

Chip-Tronics, 2019-05-28:
I found out that it doesn't work with the Raspberry pi 2B.
I bought a 2nd LCD and it did the exact same. It never worked.
Then I bought the Raspberry pi 2B+, and it was just plug and play.
I then tried my Raspberry pi 1. It worked as well.
So it was the Raspberry pi 2B. It's just not compatible.
Thanks for responding!

goodtft, 2019-07-01:
see this:
https://github.com/lcdwiki/LCD-show-ubuntu