goodtft/LCD-show #124: Can I disable the Touch screen?

sonicnick, 2019-03-07:
Screen works great, but I dont need the touchscreen can I disable that? 
Cant see anything in the config-35.txt (I have the 3.5 pi version)

goodtft, 2019-03-08:
Yes  ,you can.
The simplest solution is hardware:
Cut off the four wires on the touch pad and you don't have touch input.
You can even remove the top touch panel without affecting the display

2019-03-08 

goodtft

vinayprydotinfotech, 2020-06-03:
Can you Show or Share any video to this LCD removing Touch Portion. I am really not interested in touch.. What if I cut the track of Chip Select of xpt2046 and Hang that Chip Select to High logic ? would that do the trick ?

vinayprydotinfotech, 2020-06-03:
> Screen works great, but I dont need the touchscreen can I disable that?
> Cant see anything in the config-35.txt (I have the 3.5 pi version)

could you achieve that what you needed ? .i.e. safely disabling touch

bachoo786, 2023-12-24:
How to disable touchscreen via software?