goodtft/LCD-show #126: raspberry kernel not recognized with 2018-11-13-raspbian-stretch-full

tm2641, 2019-03-10:
I tried to use LCD35-show with Raspian Stretch but it fails.
pi@raspi:~ $ chmod -R 755 LCD-show
pi@raspi:~ $ cd LCD-show/
pi@raspi:~/LCD-show $ sudo ./LCD35-show
this is not raspberrypi kernel, no need to update touch configure, reboot

The touch function is not available therefore.
How can I update touch configure manually?
Is there a detailed description, what the program changes on the raspberry to check this manually?

Best regards Tom

nizoo91, 2019-03-10:
**Option 1:** 
you need to edit the following file in your case: LCD35-show
cd LCD-show
sudo nano LCD35-show
//edit line #18, 
if test "$nodeplatform" = "raspberrypi";then
TO
if test "$nodeplatform" = "**raspi**";then

**Option 2:**
sudo raspi-config
2 Network Options
N1 Hostname
change hostname to "raspberrypi"
and try to run the whole thing again

Good luck

tm2641, 2019-03-10:
Great, thank you so much. Now I realized that it was a script and the issue is solved. Best regards Tom

tm2641, 2019-03-10:
issue is solved

closed 2019-03-10 by tm2641