goodtft/LCD-show #130: to anyone have a 3.5 inch display doesn't work

pieeatguy, 2019-04-15:
I find out that this "MHS-3.5inch RPi Display" is a cheap-knockoff from waveshare's  3.5inch RPi LCD (C)(https://www.waveshare.com/3.5inch-rpi-lcd-c.htm), from chip on the back to spi speed they cliamed.And waveshare's driver can work fine for this.
but sadly, touch screen is still reversed :(

https://github.com/waveshare/LCD-show/issues

wally666, 2019-04-20:
Where is the problem? With calibration or what?

`
sudo nano /etc/X11/xorg.conf.d/99-calibration.conf
`

and change to:

`
Section "InputClass"
        Identifier      "calibration"
        MatchProduct    "ADS7846 Touchscreen"
        Option  "Calibration"   "277 3936 3880 268"
        Option  "SwapAxes"      "1"
EndSection
`

or read the [manual](http://www.lcdwiki.com/res/Show_Direction_and_Touch/How_to_change_display_direction-GPIO-Resistive_Touch-V1.2.pdf)

ramteen85, 2019-04-26:
there is another problem with this "cheap knock off".
when i execute ./LCD-hdmi, it doesn't boot up in hdmi.
you can see it for a little bit on the screen but it doesn't boot up in hdmi.

xuancong84, 2023-12-25:
I fully agree, the device is a piece of garbage and bullshit. No matter how I reconfigure, the touchscreen is still swapped axis. Never works right.