goodtft/LCD-show #132: Is there simple code to get touch coordinates from the touchscreen on 3.5" touch screen?

marcmerlin, 2019-04-21:
I would love to get a simple feed of coordinates from the touchscreen without having to install or use Xorg.
On arduino, I get a simple driver that outputs coordinates and a touch force number.

Is there any basic code that can do this with this touch screen in python or C?

Thanks

marcmerlin, 2019-04-22:
@goodtft is the touch driver compatible with XPT2046?
https://github.com/PaulStoffregen/XPT2046_Touchscreen
If not, is there an arduino driver for it?
Basically I'd like to do the same thing than what I did on arduino, but with your screen on rPi:
https://youtube.com/watch?v=Kvcvpdip12A

marcmerlin, 2019-04-30:
Right, but "I would love to get a simple feed of coordinates from the touchscreen without having to install or use Xorg." :)
That said, what you give is indeed reasonably easy if I have to use Xorg

texadactyl, 2019-04-30:
One only needs the bare minimum of Xorg, not a whole desktop.  You might find that Xorg has solved your current needs & future needs on evolving/multiple hardware.  I've been using Xorg in C/C++ for a number of years (since Berkeley Unix) and appreciate the level of transparency it provides.

Good luck with your project.