goodtft/LCD-show #133: LCD-hdmi breaks my operating system :(

ramteen85, 2019-04-26:
Hey everyone. I just bought my LCD screen and I am gravely disappointed.
I thought this was a problem with the manufacturer so I just went and bought another screen.

You guys don't even have the drivers working??

So now I have bought 2 screens, they are both going to be utterly useless, and hdmi won't even work, so it was better to work headless this whole time.

awesome.

Is there going to be a fix for this anytime soon?
looks like it's not compatible for kali 2019.

To everyone else:
Unless you do not use HDMI, do NOT install these drivers. they WILL break your operating system. They will work on the lcd screen but as soon as you try to boot back into hdmi you are screwed. neither the LCD screen or the HDMI will work.

Crappy piece of junk I can't believe I wasted my $$ on this

mccombs, 2019-05-16:
Yep. Right there with you.

robdejonge, 2019-05-28:
If you have anything but a default install, see https://github.com/goodtft/LCD-show/issues/140#issuecomment-496350699 ... this may help.

ramteen85, 2019-05-28:
Nope. Doesn't solve my issue for shit. The script is broken. The best solution says "pick it apart". 

How about get off your asses and update the script so people don't waste their money. Or don't sell a ripoff product. 

Unbelievable. You're selling something that doesn't even work. Fresh install or not.

robdejonge, 2019-05-28:
Yo. I'm just another user trying to be friendly. Really no need for the aggressive comments.

ramteen85, 2019-05-28:
Sorry man. I've had this thing for a while and no solution. It's still
being sold freely. Frustrating as hell. Not trying to be aggressive with
you but this is q massive fail on the manufacturers part.

robdejonge, 2019-05-28:
Agreed.