goodtft/LCD-show #135: for me, the 3.5 inch panel does not work

CuzImBisonratte, 2019-05-01:
I entered the first command and then I was told that I should log in, must I log in with my github data? or did I do something wrong?

CuzImBisonratte, 2019-05-02:
I have entered the commands now it went. But in the last command (sudo ./LCD35-show) I do not have the necessary permissions. Is there any other command, even if it is more complicated?

goodtft, 2019-05-05:
something wrong with you system.
try use Raspbian from https://www.raspberrypi.org/downloads/

If you have difficulty installing the driver, or if you still can't use the display properly after installing the driver,
Please Try our Configned images for tested.
Just need download and write the image into the Micro SD card. DO NOT need any driver installation steps.
see our wiki:
http://www.lcdwiki.com/Main_Page

2019-05-05 

goodtft

CuzImBisonratte, 2019-05-05:
I tried it again after a day, now everything works and I am very happy. Thank you for this great product goodtft.

closed 2019-05-05 by CuzImBisonratte