goodtft/LCD-show #137: The current CMakeCache.txt directory

vadimk2016, 2019-05-12:
After it runs the sudo ./LCD5-show command, it throws an error:
`CMake Error: The current CMakeCache.txt directory /home/pi/downloads/LCD-show/rpi-fbcp/build/CMakeCache.txt is different than the directory /home/pi/LCD-show/rpi-fbcp/build where CMakeCache.txt was created. This may result in binaries being created in the wrong place. If you are not sure, reedit the CMakeCache.txt CMake Error: The source "/home/pi/downloads/LCD-show/rpi-fbcp/CMakeLists.txt" does not match the source "/home/pi/LCD-show/rpi-fbcp/CMakeLists.txt" used to generate cache. Re-run cmake with a different source directory. CMake Error: The source directory "/home/pi/LCD-show/rpi-fbcp" does not exist. Specify --help for usage, or press the help button on the CMake GUI. Makefile:176: recipe for target 'cmake_check_build_system' failed make: *** [cmake_check_build_system] Error 1 LCD configrue 0`

goodtft, 2019-07-01:
I think that you should delete the directory https://github.com/lcdwiki/LCD-show.git and run the sudo ./LCD5-show command again.

goodtft, 2019-07-01:
I think that you should delete directory /home/pi/downloads/LCD-show/rpi-fbcp/build and run the sudo ./LCD5-show command again.