goodtft/LCD-show #138: LCD works but PI won't boot

mccombs, 2019-05-16:
Followed the instructions screen works but it borks my OS

Running OctoPrint - https://octoprint.org/

Anyone have a solution?

robdejonge, 2019-05-28:
If you have anything but a default install, see my comment at https://github.com/goodtft/LCD-show/issues/140#issuecomment-496350699 ... this may help.

michaelsheridan23, 2019-06-04:
Not sure if this is allow but these drivers work perfectly with this screen and are octoprint compatible: https://www.elecrow.com/wiki/index.php?title=3.5_Inch_480x320_TFT_Display_with_Touch_Screen_for_Raspberry_Pi

fschlaef, 2019-06-13:
@mccombs I am creating a better alternative here : https://github.com/fschlaef/raspi-lcd/
Would you mind giving it a try ?

goodtft, 2019-07-01:
see this
https://github.com/lcdwiki/LCD-show.git