goodtft/LCD-show #140: Horrible, horrible script. Just, horrible.

robdejonge, 2019-05-28:
This script (I ran `LCD5-show`) completely violates every concept of respect towards the device it's being run on. A list of its violations : 

- Removes `/etc/X11/xorg.conf.d/40-libinput.conf` entirely. No backup made, no message displayed, no confirmation required. Just, gone. Hope you have a backup. 
- Overwrites `/boot/config.txt` flat out. No backup made, no message displayed, no confirmation required. Original gone. If you had settings other than the default, well now you don't. 
- It attempts to copy a file into `/usr/share/X11/xorg.conf.d/99-fbturbo.conf`, but neglects to check if this directory already exists and thus fails if it doesn't, like was the case on my machine. 
- If the hostname of your device is anything other than `raspberrypi`, it assumes yours is not a Raspberry Pi and neglects to install the `xserver-xorg-input-evdev` package one assumes is needed to make it work and neglects to copy a file into `/usr/share/X11/xorg.conf.d` which may have failed anyway as the directory doesn't exist on every machine. 

The biggest failure of this script, and which rendered my (and I imagine many other) Pi's silent is the fact that it overwrites `/boot/cmdline.txt` with a directive `root=/dev/mmcblk0p2`. This will work if you are running your Pi off a standard Raspbian install run off an SD card. However, anything other than pure defaults and you're screwed. And why, why change that directive? There is no reason to change this to make the screen work. This script should have processed the `cmdline.txt` file and changed the appropriate directives where needed and left the other ones as they were. 

What an absolutely horrendous scripting job. Unbelievable.

robdejonge, 2019-05-28:
If you have a Pi that does not boot after running this script, mount your drive on another host and edit the `cmdline.txt` back to what it used to be. Hope you have a backup! I boot from an SSD, so I just changed the `rootfs=` directive back to what it was.

fschlaef, 2019-06-05:
OK, THAT is why I pulled my hair out yesterday trying to understand why HDMI wasn't working and my config was messed up.

This script is nothing short of malicious. I guess you can't expect much more from cheap Chinese crap sold on Amazon, but still.

Time to reinstall Raspbian.

Can we do something ? A fork perhaps ?

robdejonge, 2019-06-06:
Why would you have to reinstall Raspbian? Edit your `cmdline.txt` and everything should be fine. 

Yes, this repository could be forked and better scripts could be created. Let me know when it's up! 😉

fschlaef, 2019-06-06:
@robdejonge Because of overwritten files and issues I had.
I did reinstall and everything worked from the get go : HDMI, sound, keyboard, all the things that gave me trouble before work now. Maybe the fault of the script, maybe not.

ThomasIommi, 2019-06-09:
I had the same issue trying to configure my Kuman 5" LCD screen on a Kali Linux 2019.2 for a Raspberry PI B+ too... Kernel panic and stuff... the script is indeed really bad... to me the issue was on the /boot/config.txt file though.. because I've tryed some config found here: [https://www.elecrow.com/wiki/index.php?title=HDMI_Interface_5_Inch_800x480_TFT_Display](https://www.elecrow.com/wiki/index.php?title=HDMI_Interface_5_Inch_800x480_TFT_Display) and it worked.. surprisingly even the touchpad worked (kinda... not super precise).
I've followed the instruction on the step2 part and ignored EVERYTHING else.. Then I've tried again (for science!) only the configs of /boot/config.txt found in this repo and I've got kernel panic again.. 
So I'm just sticking with adding this:
`hdmi_force_hotplug=1
max_usb_current=1
hdmi_drive=1
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0
dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900
display_rotate=0`
to my default /boot/config.txt 

Hope it helps!

ThomasIommi, 2019-06-11:
After using xinput-calibrator and saving it's result in /usr/share/X11/xorg-conf.d/99-calibration.conf even the touchpad is working correctly!

fschlaef, 2019-06-11:
@ThomasIommi  @robdejonge I got to work fixing this script, we can discuss here : https://github.com/fschlaef/Raspberry-Pi-LCD-Driver/issues/1

ThomasIommi, 2019-06-13:
After some testing I changed my configuration a bit:
```
hdmi_force_hotplug=1
max_usb_current=1
hdmi_drive=2
hdmi_group=2
# 848x480 60Hz 16:9
hdmi_cvt 848 480 60 3 0 0 0
# 1920x1080 60Hz 16:9
#hdmi_cvt 1920 1080 60 3 0 0 0
dtoverlay=ads7846,cs=1,penirq=25,penirq_pull=2,speed=50000,keep_vref_on=0,swapxy=0,pmax=255,xohms=150,xmin=200,xmax=3900,ymin=200,ymax=3900
display_hdmi_rotate=0
```
- Switched to hdmi_drive=2, just because if I connect another hdmi screen that supports audio it's boring to change too many configurations and it shouldn't give any problem.
- Removed hdmi_mode, because I belive it gets overridden by hdmi_cvt anyway.
- Changed the resolution and aspect ratio; I don't know why but with 848x480 resolution and aspect ratio of 16:9 is much better then the suggested 800x480 15:9 in my case (even if the screen is 800x480).
- Changed display_rotate=0 with display_hdmi_rotate=0 because I saw display_hdmi param is deprecated.

The dtoverlay part is for the touchscreen I believe, but I didn't test it yet

fschlaef, 2019-06-14:
@ThomasIommi would you mind heading to https://github.com/fschlaef/raspi-lcd/issues/2 and post your config ? Thank you very much 👍

sc4773rm1nd, 2019-07-02:
The script also overwrites certain rasp pi 4 functions found on raspbian buster

jdimpson, 2019-09-22:
The install scripts also reboot without prompting. Super since I was in the middle of downloading data in another window.

Don't reboot. Simply print "please reboot" now.

True-Demon, 2020-01-10:
This :arrow_up:. Exactly this.

Ran into the same issues. Kernel panic every single time. What I find most malicious about this script is that it reboots your pi with no warning whatsoever or giving you any chance to address the absolute CF of errors that the script generates. In my case, I was trying to set this up on a Kali-based rpi image, and it generated hundreds of cmake errors. The cmake files _assume_ that you saved the LCD-show drivers in the `/home/pi/` directory instead of simply checking the current directory, which I find just plain baffling.

This script needs a complete re-write desperately. Every time I see a "sudo reboot" in this script, it makes me feel like someone just kicked a puppy. :angry:

lbussy, 2020-01-10:
Then submit a PR fixing it.

bmelim, 2020-07-19:
I have got mine to work and created some instructions. Also used someone else script and edited to make installing easier.

https://github.com/bmelim/raspi-lcd5-touchscreen

Let me know your thoughts