goodtft/LCD-show #143: 3.5inch RPI Display does not work on Ubuntu mate

papayalabs, 2019-06-08:
I try to install drivers in ubuntu mate ( image --> ubuntu-mate-18.04.2-beta1-desktop-armhf+raspi-ext4.img) with ./LCD35-show and there is no way to run the LCD Display and break the system.

I try to install the image --> MPI3501-3.5inch-Ubuntu-18.04.2.img, the LCD works but when I try to switch to HDMI with ./LCD-hdmi it does not work and break the system again.

Thanks,

fschlaef, 2019-06-13:
@papayalabs Unfortunately this script is very poorly coded and erases several system config files without warning, as explained here : https://github.com/goodtft/LCD-show/issues/140

I am creating a better alternative here : https://github.com/fschlaef/raspi-lcd/ 
Would you mind giving it a try ?

sceic, 2019-06-24:
> @papayalabs Unfortunately this script is very poorly coded and erases several system config files without warning, as explained here : #140
> 
> I am creating a better alternative here : https://github.com/fschlaef/raspi-lcd/
> Would you mind giving it a try ?

It's worked ,MHS35.

goodtft, 2019-07-01:
see this:
https://github.com/lcdwiki/LCD-show-ubuntu
for LCD35-show

closed 2019-11-26 by goodtft