goodtft/LCD-show #144: Driver does not compile on Kali 2019.2

abujed, 2019-06-15:
Fails to compile and bricks OS - trace below...
HELP!?

root@kali:~/Downloads/LCD-show# ./MHS35-show 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.20.1-2).
The following additional packages will be installed:
  cmake-data librhash0 libuv1
Suggested packages:
  cmake-doc ninja-build
The following NEW packages will be installed:
  cmake cmake-data librhash0 libuv1
0 upgraded, 4 newly installed, 0 to remove and 1 not upgraded.
Need to get 4,747 kB of archives.
After this operation, 24.3 MB of additional disk space will be used.
Get:1 http://archive.linux.duke.edu/kalilinux/kali kali-rolling/main arm64 cmake-data all 3.13.4-1 [1,476 kB]
Get:2 http://archive.linux.duke.edu/kalilinux/kali kali-rolling/main arm64 librhash0 arm64 1.3.8-1 [121 kB]
Get:3 http://archive.linux.duke.edu/kalilinux/kali kali-rolling/main arm64 libuv1 arm64 1.24.1-1 [105 kB]
Get:4 http://archive.linux.duke.edu/kalilinux/kali kali-rolling/main arm64 cmake arm64 3.13.4-1 [3,046 kB]
Fetched 4,747 kB in 7s (653 kB/s)                                                                                                                                        
Selecting previously unselected package cmake-data.
(Reading database ... 254665 files and directories currently installed.)
Preparing to unpack .../cmake-data_3.13.4-1_all.deb ...
Unpacking cmake-data (3.13.4-1) ...
Selecting previously unselected package librhash0:arm64.
Preparing to unpack .../librhash0_1.3.8-1_arm64.deb ...
Unpacking librhash0:arm64 (1.3.8-1) ...
Selecting previously unselected package libuv1:arm64.
Preparing to unpack .../libuv1_1.24.1-1_arm64.deb ...
Unpacking libuv1:arm64 (1.24.1-1) ...
Selecting previously unselected package cmake.
Preparing to unpack .../cmake_3.13.4-1_arm64.deb ...
Unpacking cmake (3.13.4-1) ...
Setting up libuv1:arm64 (1.24.1-1) ...
Setting up librhash0:arm64 (1.3.8-1) ...
Setting up cmake-data (3.13.4-1) ...
Setting up cmake (3.13.4-1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.28-10) ...
Cloning into 'rpi-fbcp'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 61 (delta 3), reused 5 (delta 1), pack-reused 52
Unpacking objects: 100% (61/61), done.
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /root/Downloads/LCD-show/rpi-fbcp/build
Scanning dependencies of target fbcp
[ 50%] Building C object CMakeFiles/fbcp.dir/main.c.o
/root/Downloads/LCD-show/rpi-fbcp/main.c:9:10: fatal error: bcm_host.h: No such file or directory
 #include <bcm_host.h>
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/fbcp.dir/build.make:63: CMakeFiles/fbcp.dir/main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/fbcp.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
install: cannot stat 'fbcp': No such file or directory
this is not raspberrypi kernel, no need to update touch configure, reboot
Connection to <ip address> closed by remote host.
Connection to <ip address> closed.

I would post other pertinent system info but it's bricked :-(

runopsrun, 2019-06-21:
Same issue here and a bricked system.

root@kali:~/LCD-show# ./MHS35-show
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version (1:2.20.1-2).
The following additional packages will be installed:
  cmake-data librhash0 libuv1
Suggested packages:
  cmake-doc ninja-build
The following NEW packages will be installed:
  cmake cmake-data librhash0 libuv1
0 upgraded, 4 newly installed, 0 to remove and 1 not upgraded.
Need to get 4747 kB of archives.
After this operation, 24.3 MB of additional disk space will be used.
Get:1 http://kali.download/kali kali-rolling/main arm64 cmake-data all 3.13.4-1 [1476 kB]
Get:2 http://kali.download/kali kali-rolling/main arm64 librhash0 arm64 1.3.8-1 [121 kB]
Get:3 http://kali.download/kali kali-rolling/main arm64 libuv1 arm64 1.24.1-1 [105 kB]
Get:4 http://kali.download/kali kali-rolling/main arm64 cmake arm64 3.13.4-1 [3046 kB]
Fetched 4747 kB in 1s (4126 kB/s)
Selecting previously unselected package cmake-data.
(Reading database ... 254704 files and directories currently installed.)
Preparing to unpack .../cmake-data_3.13.4-1_all.deb ...
Unpacking cmake-data (3.13.4-1) ...
Selecting previously unselected package librhash0:arm64.
Preparing to unpack .../librhash0_1.3.8-1_arm64.deb ...
Unpacking librhash0:arm64 (1.3.8-1) ...
Selecting previously unselected package libuv1:arm64.
Preparing to unpack .../libuv1_1.24.1-1_arm64.deb ...
Unpacking libuv1:arm64 (1.24.1-1) ...
Selecting previously unselected package cmake.
Preparing to unpack .../cmake_3.13.4-1_arm64.deb ...
Unpacking cmake (3.13.4-1) ...
Setting up libuv1:arm64 (1.24.1-1) ...
Setting up librhash0:arm64 (1.3.8-1) ...
Setting up cmake-data (3.13.4-1) ...
Setting up cmake (3.13.4-1) ...
Processing triggers for man-db (2.8.5-2) ...
Processing triggers for libc-bin (2.28-10) ...
Cloning into 'rpi-fbcp'...
remote: Enumerating objects: 9, done.
remote: Counting objects: 100% (9/9), done.
remote: Compressing objects: 100% (8/8), done.
remote: Total 61 (delta 3), reused 5 (delta 1), pack-reused 52
Unpacking objects: 100% (61/61), done.
-- The C compiler identification is GNU 8.3.0
-- The CXX compiler identification is GNU 8.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /root/LCD-show/rpi-fbcp/build
Scanning dependencies of target fbcp
[ 50%] Building C object CMakeFiles/fbcp.dir/main.c.o
/root/LCD-show/rpi-fbcp/main.c:9:10: fatal error: bcm_host.h: No such file or directory
 #include <bcm_host.h>
          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/fbcp.dir/build.make:63: CMakeFiles/fbcp.dir/main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:73: CMakeFiles/fbcp.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
install: cannot stat 'fbcp': No such file or directory
this is not raspberrypi kernel, no need to update touch configure, reboot

bseb, 2019-06-24:
+1 , same symptoms as above

fschlaef, 2019-06-25:
@abujed  @runopsrun  @bseb  this script has many issues, and overwrites system config as explained here : https://github.com/goodtft/LCD-show/issues/140

I have created a better alternative here that should work fine as a replacement for `MHS35-show` : https://github.com/fschlaef/raspi-lcd

Would you mind giving it a try ? At the very least, it won't brick your system :)

abujed, 2019-06-26:
Hello Francois,
Thank you for your efforts!

The good news is, the process did not brick this installation!

The bad news is... (and I may have just gotten bad hardware :-/) that I
still have a white screen.  After the installation/reboot, the signal
stopped going out the hdmi port (which I'm guessing is expected and fine)
so obviously *something* changed :-D.

Are you the vendor? or related to the vendor? Or just a good Internet
citizen trying to help?

Thanks!
Abujed

abujed, 2019-06-26:
Hi Francois,
I have more news...
Having nothing to lose, I connected a different vendor's screen (
https://www.amazon.com/gp/product/B01N447AEY/) on this rpi/kali
installation (with your driver).

The driver from that vendor also did not work so great.  It displayed at
too high of a resolution to be useful, and the touch screen did not work at
all.

Your drive set the resolution perfectly!  And the touch screen works!

However, it appears that the touch is somehow inverted, (pressing on the
left hand side of the screen affects things on the right, touching the top
activates things on the bottom) can this be adjusted for/calibrated
somewhere?

Thanks so much!!!
Abujed

runopsrun, 2019-06-26:
Same, white screen but no brick

abujed, 2019-06-26:
Sorry for the spam, I meant to include this earlier in case it might be
helpful - from the installation of your driver (sanitized and with spaces
for readability):

root@kali:~# cat /etc/*release
DISTRIB_ID=Kali
DISTRIB_RELEASE=kali-rolling
DISTRIB_CODENAME=kali-rolling
DISTRIB_DESCRIPTION="Kali GNU/Linux Rolling"
PRETTY_NAME="Kali GNU/Linux Rolling"
NAME="Kali GNU/Linux"
ID=kali
VERSION="2019.2"
VERSION_ID="2019.2"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="https://www.kali.org/"
SUPPORT_URL="https://forums.kali.org/"
BUG_REPORT_URL="https://bugs.kali.org/"

root@kali:~# uname -a
Linux kali 4.19.29-Re4son-v8+ #6 SMP PREEMPT Wed Mar 27 00:15:50 UTC 2019
aarch64 GNU/Linux

root@kali:~/Downloads# git clone https://github.com/fschlaef/raspi-lcd
Cloning into 'raspi-lcd'...
remote: Enumerating objects: 25, done.
remote: Counting objects: 100% (25/25), done.
remote: Compressing objects: 100% (20/20), done.
remote: Total 25 (delta 6), reused 11 (delta 3), pack-reused 0
Unpacking objects: 100% (25/25), done.

root@kali:~/Downloads/raspi-lcd# ./LCD_480_320.sh
evdev already installed. Nothing to do.
Calibration config file doesn't exists. Creating it ...
Saving /boot/config.txt current values and writing new config ...
Install done. A reboot is needed to start using your touchscreen
Reboot now ? [Y/N/q] : y
Rebooting ...

Last login: Wed Jun 26 19:58:04 2019 from <somewhere>
root@kali:~# uptime
 20:06:46 up 5 min,  1 user,  load average: 0.09, 0.12, 0.06
root@kali:~# uptime
 20:12:04 up 10 min,  1 user,  load average: 0.00, 0.08, 0.07
root@kali:~#

fschlaef, 2019-06-27:
Hello @abujed and thank you for testing my script !

I am not the vendor neither affiliated to it, I am indeed just trying to help :)

I have not tested on Kali Linux yet so that might be the cause of the white screen you are encountering.
About the inverted X/Y axis, try this :

- Open /usr/share/X11/xorg.conf.d/99-calibration.conf
- Change `SwapAxes` to `"0"`
- Reboot

For further issues / questions, would you mind opening them on my repo ? I don't want to have problems 😄 

https://github.com/fschlaef/raspi-lcd/issues

Thank you very much !

abujed, 2019-06-27:
> <https://github.com/fschlaef/raspi-lcd/issues> Thank you very much ! —You

Thanks for the info Francois.
I'm happy to open that question on your repo if it would help!?
Thanks

abujed, 2019-06-27:
Hi Francois,
I made that change, it's still off the mark, but... in a different way.
:-/ Moving left goes down, right up, etc.

I noticed in the 99-calibration.conf you spelled out this model - ADS7846
which is not what I'm using it with since the other screen isn't working at
all.  Do I need to track down a model number (I looked on the back but I
couldn't find anything), will dmesg or some other utility give the
nomenclature to me (if you even need it).

Also, do you want me to move this entire line of questioning to your repo?

Thanks!

goodtft, 2019-06-29:
see this:
https://github.com/lcdwiki/LCD-show-kali

2019-06-29 

goodtft 

Connection to closed by remote host.
Connection to closed.

goodtft, 2019-06-29:
see this:
https://github.com/lcdwiki/LCD-show-kali

2019-06-29 

goodtft

goodtft, 2019-06-29:
see this:
https://github.com/lcdwiki/LCD-show-kali

2019-06-29 

goodtft 

https://github.com/fschlaef/raspi-lcd/issues

DonCanDo, 2023-12-29:
> @abujed @runopsrun @bseb this script has many issues, and overwrites system config as explained here : #140
>
> I have created a better alternative here that should work fine as a replacement for [...]

This  made my day. Thanks man