goodtft/LCD-show #146: Unable to rotate mhs35 display

XOIIO, 2019-06-22:
Hi, I'm unable to rotate the mhs35 display by adding dtoverlay=mhs35:rotate=270 into /boot/config.txt or in /boot/mhs35.txt, how can I do this?

goodtft, 2019-06-29:
see our new github,it can run with params:
https://github.com/lcdwiki/LCD-show

Run:
sudo ./MHS35-show 90

2019-06-29 

goodtft

werty1st, 2019-07-14:
`./LCD35-show 270
dpkg: dependency problems prevent removal of xserver-xorg-input-evdev:
 xserver-xorg depends on xserver-xorg-input-all | xorg-driver-input; however:
  Package xserver-xorg-input-all is not installed.
  Package xorg-driver-input is not installed.
  Package xserver-xorg-input-evdev which provides xorg-driver-input is to be removed.

dpkg: error processing package xserver-xorg-input-evdev (--purge):
 dependency problems - not removing
Errors were encountered while processing:
 xserver-xorg-input-evdev
./LCD35-show: line 36: ./.have_installed: Permission denied
2019
need to update touch configuration
Selecting previously unselected package xserver-xorg-input-evdev.
(Reading database ... 68489 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-input-evdev_2.10.5-1_armhf.deb ...
Unpacking xserver-xorg-input-evdev (1:2.10.5-1) over (1:2.10.5-1) ...
Setting up xserver-xorg-input-evdev (1:2.10.5-1) ...
Processing triggers for man-db (2.8.5-2) ...

./rotate.sh: line 50: [: =: unary operator expected
./rotate.sh: line 65: [: =: unary operator expected
Invalid output type
reboot now`

goodtft, 2019-07-18:
> ```shell
> ./LCD35-show 270
> dpkg: dependency problems prevent removal of xserver-xorg-input-evdev:
>  xserver-xorg depends on xserver-xorg-input-all | xorg-driver-input; however:
>   Package xserver-xorg-input-all is not installed.
>   Package xorg-driver-input is not installed.
>   Package xserver-xorg-input-evdev which provides xorg-driver-input is to be removed.
> 
> dpkg: error processing package xserver-xorg-input-evdev (--purge):
>  dependency problems - not removing
> Errors were encountered while processing:
>  xserver-xorg-input-evdev
> ./LCD35-show: line 36: ./.have_installed: Permission denied
> 2019
> need to update touch configuration
> Selecting previously unselected package xserver-xorg-input-evdev.
> (Reading database ... 68489 files and directories currently installed.)
> Preparing to unpack .../xserver-xorg-input-evdev_2.10.5-1_armhf.deb ...
> Unpacking xserver-xorg-input-evdev (1:2.10.5-1) over (1:2.10.5-1) ...
> Setting up xserver-xorg-input-evdev (1:2.10.5-1) ...
> Processing triggers for man-db (2.8.5-2) ...
> 
> ./rotate.sh: line 50: [: =: unary operator expected
> ./rotate.sh: line 65: [: =: unary operator expected
> Invalid output type
> reboot now
> ```

You should use "sudo ./LCD35-show 270"

closed 2019-11-26 by goodtft