goodtft/LCD-show #152: error processing package xserver-xorg-input-evdev (--install)

athulsnair, 2019-07-09:
Hello,   

  I'm trying to add a LCD touch screen to Octoprint.
I bought a 5 inch LCD and came with a driver CD. The CD comes with instructions, and uses this driver. When I enter _sudo ./LCD5-show_ it throws bunch of errors and reboot

`
pi@octopi:~ $ ls
LCD-show  mjpg-streamer  OctoPrint  oprint  scripts
pi@octopi:~ $ sudo chmod -R 755 LCD-show
[sudo] password for pi:
pi@octopi:~ $ sudo chmod -R 755 LCD-show
pi@octopi:~ $ cd LCD-show/
pi@octopi:~/LCD-show $ sudo ./LCD5-show
(Reading database ... 44329 files and directories currently installed.)
Removing xserver-xorg-input-evdev (1:2.10.5-1) ...
dpkg: warning: while removing xserver-xorg-input-evdev, directory '/usr/share/X11/xorg.conf.d' not empty so not removed
Processing triggers for man-db (2.7.6.1-2) ...
2019
need to update touch configuration
Selecting previously unselected package xserver-xorg-input-evdev.
(Reading database ... 44316 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-input-evdev_2.10.5-1_armhf.deb ...
Unpacking xserver-xorg-input-evdev (1:2.10.5-1) ...
Processing triggers for man-db (2.7.6.1-2) ...
dpkg: dependency problems prevent configuration of xserver-xorg-input-evdev:
 xserver-xorg-input-evdev depends on libevdev2 (>= 0.9.1); however:
  Package libevdev2 is not installed.
 xserver-xorg-input-evdev depends on libmtdev1 (>= 1.1.0); however:
  Package libmtdev1 is not installed.
 xserver-xorg-input-evdev depends on xorg-input-abi-24; however:
  Package xorg-input-abi-24 is not installed.
 xserver-xorg-input-evdev depends on xserver-xorg-core (>= 2:1.18.99.901); however:
  Package xserver-xorg-core is not installed.

dpkg: error processing package xserver-xorg-input-evdev (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 xserver-xorg-input-evdev
reboot now

`
These are the instructions came with the CD and looks similar to your git hub page
[How to install the LCD driver (En).pdf](https://github.com/goodtft/LCD-show/files/3371770/How.to.install.the.LCD.driver.En.pdf)
[5inch HDMI Display User Manual(EN).pdf](https://github.com/goodtft/LCD-show/files/3371771/5inch.HDMI.Display.User.Manual.EN.pdf)

athulsnair, 2019-07-09:
I tried to install evrything from start, after formatting SD card.
Same error shoed. Tried to manullay install those missing packages. I have not used Linux based system before, so don't know whether my steps are right.

I did like this
Tried 
```
sudo apt-get update
sudo apt-get install libevdev2
```

In one of these installations, it showed some warning and recommend to do `sudo apt --fix-broken install` , and after that I was able to install the driver.

I'm using this [video ](https://www.youtube.com/watch?v=rqy3dVZ4JH4)as reference. Letter were still small even after driver installation, unlike what's shown  in the video. So I doubt whether it's properly installed

Here's the image of LCD 

[image]

athulsnair, 2019-07-09:
How to driver was successfully installed or not ?

goodtft, 2019-07-18:
You should install the libraries that are not installed and execute the commands  sudo ./LCD5-show

athulsnair, 2019-07-25:
Did than and worked, sorry for the late reply

closed 2019-07-25 by athulsnair