goodtft/LCD-show #159: Doesn't work with raspbian buster

kubyk12, 2019-08-21:
Hi! I have an issue with my display. I use the drivers ,activate it and the LCD is still white after rebooting. I INSTALLED THE OS 7 TIMES AND DOESN'T WORK! FIX IT!!

goodtft, 2019-08-24:
which LCD display?
and your raspbian version is?

| |
goodtft
|
|
goodtft@163.com
|
签名由网易邮箱大师定制

kubyk12, 2019-08-26:
US $11 | New Original 3.5" LCD TFT Touch Screen Display for Raspberry Pi 3 Model B Board
https://s.click.aliexpress.com/e/KX7zyPNSb
And I are raspbian buster

goodtft, 2019-08-27:
https://item.taobao.com/item.htm?spm=a1z10.5-c.w4002-822319412.13.64355e80RLkWK4&id=579493092990

| |
goodtft
|
|
goodtft@163.com
|
签名由网易邮箱大师定制

https://s.click.aliexpress.com/e/KX7zyPNSb

liverun, 2019-09-13:
I was able to make the LCD screen work with Raspbian Buster by performing a manual installation.
This guide for example can help: https://www.root42.de/blog/?p=6

I use something like this for my Xorg conf:
```
Section "InputClass"
    Identifier "Touchscreen"
    Driver "evdev"
    MatchProduct "ADS7846"
    Option "EmulateThirdButton" "1"
    Option "EmulateThirdButtonTimeout" "750"
    Option "EmulateThirdButtonThreshold" "50"
EndSection
```
This will require package evdev. If it's not already installed on your system:
`sudo apt-get install libevdev2 xserver-xorg-input-evdev`

I also had to add some calibration directives because the cursor was moving in the opposite direction. YMMV.

lyneca, 2019-09-14:
Nope, after doing that I just get a white screen. I honestly think it's a faulty screen - I'm able to get a 5" HDMI one working fine. I've already ordered a new one from a far more reliable source than banggood.com, so we'll see how that goes.

goodtft, 2019-09-21:
OK,waiting for your good news

lyneca, 2019-09-21:
After it arrived, I tried LCD-show and while it did work, I got abysmal framerate (~5fps). I then discovered the excellent [`fbcp-ili9341`](https://github.com/juj/fbcp-ili9341/) library which worked perfectly (albeit lacking touchscreen support) - giving me an excellent framerate. `fbcp-ili9341` also has the benefit of not overwriting system files without making any backups, and drastically reduced the time I had to spend on reformatting SD cards after they were broken by LCD-show.

rafikFares, 2019-09-29:
i confirm after flashing raspbian buster on my pi 3B+, the screen work perfectly but after i do sudo apt update and sudo apt upgrade, after reboot the screen stay white !
screen : 3.5" LCD TFT Touch Screen Display
raspberry : pi 3B+
kernel version : 4.19.57-v7+ #1244

ikohler, 2019-10-06:
Having issues as well. Screen just stays white with latest buster image:
screen : 3.5" LCD TFT Touch Screen Display
Raspberry: Raspberry Pi 3 Model B Rev 1.2
kernel: 4.19.75-v7+ #1271

goodtft, 2019-10-09:
Use our image for test.
http://www.lcdwiki.com/3.5inch_RPi_Display

| |
goodtft
|
|
goodtft@163.com
|
签名由网易邮箱大师定制

ikohler, 2019-10-09:
Hi goodtft,

I have tried the image already, same issue. Display stays white :-(

obzerving, 2019-10-10:
I also downloaded the image at http://www.lcdwiki.com/3.5inch_RPi_Display onto a Raspberry Pi 3B+ and the display stays white.

goodtft, 2019-11-26:
> Hi goodtft,
> 
> I have tried the image already, same issue. Display stays white :-(

Is the screen which you're using our 3.5inch RPI screen? it's not mhs35.

worstenbrood, 2019-12-12:
I have the same issue with raspbian and some screen from banggood, only white screen

[image]

codesel, 2020-03-01:
Currently experiencing the same issues with the MPU5001 capacitive screen while using the 
Raspberry Pi Zero W. My screen works and displays items correctly, as well as; changes properly when I made config changes as expected. The touchscreen feature however is not working.

Solutions (attempted):
1. Utilizing the image provided by your documentation:
 MPI5001-5inch-B-800x480-2020-02-13-raspbian-buster.7z
2. Following along in this guide while using the most recent version of Raspbian Buster (4.19): http://www.lcdwiki.com/res/MPI5001/EN/How_to_install_the_LCD_driver_V1.2.pdf
3. Finally, I have a RPI3B+, using Raspbian Jessie, after testing; looks like the screen still works on that version (touchscreen capability).

Please let me know if there is any additional testing I can do to provide more information.

daniel4wong, 2020-03-31:
> I was able to make the LCD screen work with Raspbian Buster by performing a manual installation.
> This guide for example can help: https://www.root42.de/blog/?p=6
>
> I use something like this for my [...]

WOW, my LCD screen is MPI3508, follow exact the steps in your link and magic happens!

kubyk12, 2020-04-05:
Hello. I recently found ,that my display has Brand named MINGYUANDINGYE. Raspberry Pi 3.5" display 
Raspbian Buster
Can you help me?

BingHengwlm, 2020-05-18:
I use kali linux, but it can't run. When I install it, my pi will reboot and can't output anything on my hdmi screem.
My pi is pi4b

lyneca, 2020-05-18:
The script here is [awful](https://github.com/goodtft/LCD-show/issues/140) and honestly should not be used.

I believe in that linked issue, some other people have forked the repo and are attempting to rewrite it, but in the meantime if you're not interested in using the touch screen portion of your screen I'd recommend [this script](https://github.com/juj/fbcp-ili9341) instead. It's a high-performance driver that has the benefit of not overwriting files in /boot without making any backups, and it worked perfectly with my screen.

iztoogood, 2020-08-16:
to manage to make it work again, i had to rollback the kernel
https://www.raspberrypi.org/forums/viewtopic.php?p=1706058&sid=726d26c8f4da37554e751adb26aff36c#p1706058
`sudo rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2`

mikeblakeuk, 2020-09-03:
Please can we get a fix / guide for this.
Looking at using the Pi 4 B with USB driver for the OS and LCD 3.5

mohamnag, 2021-02-22:
@lyneca can you tell us which configuration did you use for these no-name LCDs with that driver fbcp-ili9341? the best I get is some flickery output!

HyGy, 2021-07-08:
Finally I can make it work by following the MP3804-show script from line to line and what is exist and can do I created with hand. And now it is working with 2021-05-07-raspios-buster-armhf.

closed 2021-07-09 by Fiktx