goodtft/LCD-show #162: Inversed colors

0x77dev, 2019-10-15:


alveola, 2019-10-19:
My screen also has inverted colors:
Black <> White
Blue <> Cyan
Green <> Magenta
Red <> Yellow
In the image, you see the same screen sample from vnc (right, correct colors) en the RPi (left, inverted colors)
Does anyone have any hint to solve this?
[image]

0x77dev, 2019-10-19:
I 

> My screen also has inverted colors:
> Black <> White
> Blue <> Cyan
> Green <> Magenta
> Red <> Yellow
> In the image, you see the same screen sample from vnc (right, correct colors) en the RPi (left, inverted colors)
> Does anyone have any hint to solve this?
> [image]

I have the same issue

goodtft, 2019-11-26:
What type of screen do you use? 
How do you install the driver?

vishaluae, 2019-12-22:
Have the same issue. Using a 3.5" RPI screen.

vishaluae, 2019-12-22:
@goodtft Is there something that we would need to edit in /boot/config.txt of a Raspberry Pi?

mvallerie, 2020-03-13:
Same issue here on `MPI3501`.

0x77dev, 2023-06-19:
The issue hasn't been updated in a long time.

closed 2023-06-19 by 0x77dev