goodtft/LCD-show #163: display works but not touch input

saehay, 2019-10-20:
Hello,

I have connected and installed the touchscreen and drivers (LCD35) per your instruction manual. The display is working and shows my GUI desktop but I can no longer use my wired keyboard AND the touch input for the screen doesn't work.

idamage, 2019-10-21:
i have the same issue.

TheRealVira, 2019-10-30:
Same issue here too on the LCD5-show version. I am trying to get Octoprint with touchscreen to run.

TheRealVira, 2019-10-30:
@goodtft

dhaneshk, 2019-11-16:
i have the same problem. display works fine, keyboard and mouse works fine. The touch doesn't. i tried with the provided stylus and fingers.

goodtft, 2019-11-26:
Please check that the touch IC XPT2046 pins is shorted or not.
You can try to reboot the raspberry pi.

just4give, 2020-03-04:
Hi,
I am having same problem. Tried rebooting few times. But I don't know how to make sure XPT2046 is shorted or not. Can anyone guide me please?

sonicx10, 2020-04-04:
Hi,
i use this manual. https://www.root42.de/blog/?p=6
I have successfully resolved.

just4give, 2020-04-04:
Thanks @sonicx10 ... I will give it a try ...

DamageHero, 2020-04-04:
> Thanks @sonicx10 ... I will give it a try ...

can you tell me if it worked? i sent returned mine already.

paolocolourblind, 2020-06-15:
Nope. did not work on mine. Is there a test to determine if this is a kernel issue or hardware?

bmelim, 2020-07-19:
I have got mine to work and created some instructions. Also used someone else script and edited to make installing easier. 

https://github.com/bmelim/raspi-lcd5-touchscreen

srianbury, 2020-08-19:
Was having the same issue on RPI 3B+, but it's working now.
1. I first ran the MHS35B-show which displayed on the screen but the touch screen didnt work
2. I tried the script @bmelim provided and neither the display nor the touch screen worked
3. Tried MHS35-show and display and touchscreen are now working

fmaglia, 2020-10-22:
I have a similar problem.
The keyboard is working, but not the touchscreen.
I used the preparred iso, available on the tutorial website.
I installed it on a raspberry pi 3b+.
Can anyone try to help me to solve this problem?