goodtft/LCD-show #164: VNC Remote Connection/HDMI: Unable to scale

jde1126, 2019-10-22:
The HDMI output ends up being only the headless Command-Line view.

There is no way to change the resolution for VNC connections, If I accidentally click "VNC" on the top nav bar via the LCD, I have to pull the power and reboot to do anything on the screen...

How can I go about having a usable resolution? Changing in the settings does not properly apply.

jde1126, 2019-10-22:
Solution:
`chmod -R 755 LCD-show 
cd LCD-show/ 
sudo ./LCD-hdmi`

I recommend creating two scripts on the desktop to switch back and forth!

goodtft, 2019-10-23:
Thanks,we will think about it.

| |
goodtft
|
|
goodtft@163.com
|
签名由网易邮箱大师定制

avalonv, 2019-11-10:
This should be pinned. Spent a few hours trying to find a way to solve this, it was not included in the manual.

What's the point of having a repository on github if there's only one person and they don't reply to issues or answer questions?

frankc60, 2019-11-10:
Great to find this post, very helpful!

goodtft, 2019-11-26:
I'm sorry, I don't understand what you're saying.