goodtft/LCD-show #166: my unbranded 3.5inch RPi TFT LCD doesn't work at all !!

ArianGhafouri, 2019-11-01:
Hi guys, I have recently bought a 3.5inch RPi TFT LCD and a Raspberry pi B3+

but i cannot set them up, I have used all of the methods and drivers, but there was no achievement!

If I want to describe LCD's appearance, on rear side of the LCD is written "3.5inch RPi Display" with big font, and there is also CE and RoHS logos and we've got "480x320 pixel" and "XPT2046 touch controller" with a smaller font under it.

I'am also using the Raspbian Buster with desktop and recommended software (2019-09-26/KV:4.19)

I will be really thankful if anybody can help me.

aaronlynn155, 2019-11-19:
What command have you tried executing after you have downloaded the repository?

b-cass, 2019-11-24:
hi! i am also getting the same problem after doing the whole process

b-cass, 2019-11-24:
sudo ./LCD35-show

goodtft, 2019-11-26:
If you use the 3.5inch RPi TFT LCD, you should execute the following command:

sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show

After the driver is installed, Is it a white screen?
Is there an error log during installation?

lbussy, 2019-11-26:
I have the screen [purchased here](https://smile.amazon.com/gp/product/B07WQW6H9S0).  I am using Raspbian Buster and have had no issues.  I followed these instructions which were on the Amazon item page:

```
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./MHS35-show
```

Note:  I think that a fresh Raspbian Lite installation does not have Git, in that case do this step first:

```
sudo apt install git
```

wsr13990, 2020-05-05:
I also have the same issue. Only getting white screen after doing all those step