goodtft/LCD-show #168: Doesn't seem to work on the latest version of raspbian buster.  (3.97 inch)

JamieBerghmans, 2019-11-07:
I've read a few posts regarding this problem and am now experiencing the same problem (white screen no matter what I do on a fresh installation). Any possibility you could fix this in a future version or am I forced to use an old raspbian version?

Thanks in advance!

alex4men, 2019-11-22:
Tried it today with the most recent raspbian buster and it worked fine.

goodtft, 2019-11-26:
> I've read a few posts regarding this problem and am now experiencing the same problem (white screen no matter what I do on a fresh installation). Any possibility you could fix this in a future version or am I forced to use an old raspbian version?
> 
> Thanks in advance!

What type of screen do you use? What type of raspberry pi do you use?

lbussy, 2019-11-26:
No issues using Raspbian Buster dated 09-26 following [these instructions](https://github.com/goodtft/LCD-show/issues/166#issuecomment-558626138).