goodtft/LCD-show #170: Some Problem

Cyorageeee, 2019-11-16:
Hey,there.  
There are some problem on buster.  
My display is 3.5 inch.  
When I installed drivers,my display is not white but it can't touch.  
What happened?
My raspbian is released on 2019-9-26,Buster.

dhaneshk, 2019-11-16:
same here. LCD works, touch doesn't

mutlucan96, 2019-11-18:
Same. It was working fine until today, probably because of an update.

mutlucan96, 2019-11-18:
Now it works but doesn't click. Just moves the pointer.

goodtft, 2019-11-26:
Please check that the touch IC XPT2046 pins is shorted or not.

lbussy, 2019-11-26:
Also using Raspbian Buster dated 09-26, no issues following [these instructions](https://github.com/goodtft/LCD-show/issues/166).

Cyorageeee, 2019-11-28:
> Please check that the touch IC XPT2046 pins is shorted or not.

Emm... Which pin is it?

paolocolourblind, 2020-06-15:
I have the same issue too. is there a fix for this now?