goodtft/LCD-show #174: Turn off screen

SilasMueller, 2019-11-27:
Hi,
is it possible, somehow, to turn the screen off while it is not used?

lbussy, 2019-11-27:
You mean like a screensaver?

SilasMueller, 2019-11-27:
A screensaver would be one solution, yes. The best solution would be some kind of on/off switch that actually cuts the power of the screen. I dont know if this is possible to achieve with a software solution.

noloader, 2019-12-08:
+1. I'm working on a small appliance. We need to turn the display off after about 5 minutes, and re-power the display if there is motion. (Similar to a tablet).

fuji246, 2020-03-08:
Can we get this prioritized?  need to power off the screen in the night to save power. The official screen provides this. https://github.com/linusg/rpi-backlight

DanielYWoo, 2020-11-05:
I have an embedded screen with my 3d printer, and I really hope the screen can be controlled by the power management system of Pi.

vifeng, 2022-08-26:
Hi, any progress of some sort on this ? Did anyone found a solution ?

nathanielng, 2022-09-14:
I am using xscreensaver and it does help to dim the screen, but it does not turn off the backlight.