goodtft/LCD-show #177: Request for Documentation Regarding the LCD Displays

daujerrene, 2019-12-18:
Hello,
I am trying to find the implementation details and datasheets regarding these LCD displays, more particularly the [3.2" RPi Display](https://github.com/goodtft/LCD-show#32-rpi-display-mpi3201) ([LCD-wiki link](http://www.lcdwiki.com/3.2inch_RPi_Display)) in order to write a library or program for directly reading and writing to the screen without the Xorg overhead.
From what I can see in the ["LCD32-show" script](https://github.com/goodtft/LCD-show/blob/master/LCD32-show), and the decompiled dtb file as mentioned in the script, the device controller seems to be an Ilitek [9340](https://cdn-shop.adafruit.com/datasheets/ILI9340.pdf) or [9341](https://cdn-shop.adafruit.com/datasheets/ILI9341.pdf). Is this correct?

Secondly, the pin 15 of the display is labeled as "LCD instruction control, Instruction/Data Register selection" in the LCD wiki page. In the [2.4inch RPi Display For RPi 3A+](http://www.lcdwiki.com/2.4inch_RPi_Display_For_RPi_3A+) page, it is labeled additionally with "low level is instruction, high level is data." Is it the same case for the 3.2" LCD? (If they use the same controller then that should be the case, I think). What Instructions does it accept? Is it the same as those listed in the section 8 of the ILI9341 datasheet?

I am new to embedded systems programming and driver programming so I apologise for my lack of knowledge.

goodtft, 2020-05-30:
Yes,you are right, it is ILI9341. it is the same case for 3.2" LCD. The Instructions is in the ILI9341 datasheet.

daujerrine, 2020-06-29:
I see. Thank you.