goodtft/LCD-show #179: Can't seem to boot to cli

JonZso, 2019-12-30:
I'm using Pi4 w/ 4GB Ram.

I want to boot to cli as desktop isn't as useful as i wanted.
I'm using:
MHS-3.5” RPi Display(MHS3528)

Just wanted to use command line instead of gui

dr-s-mcgraw, 2020-01-19:
@JonZso So apologies if this isn't the issue you're seeing, (i'm on a 3B+) but I noticed when I set it to boot straight to console it would freeze before it gave me a prompt. Turns out as part of the installation process it launches the program `fbcp` via `/etc/rc.local`. All I had to do to fix it was edit the file via `sudo nano /etc/rc.local` and commenting it out by adding a # before the line `fbcp &`.

In my (VERY) limited testing it doesn't seem to affect anything if you're only booting to a console. I'm sure there's a far more elegant way to fix this but I feel as if with how responsive the developers have been and how poor the quality of the scripting is in general for this I don't think it'd be worth anyone's time to do a PR fixing it. 

(NOTE: If you have no other access to the pi except through the screen, you *can* also switch to another console by hitting `ctrl+alt+F2` (or F3, F4, etc.) This only seems to affect TTY1. From there you can straight-up just kill the process as well to get TTY1 back.)

JonZso, 2020-02-20:
Thanks @dr-s-mcgraw 
i will give it a check and let you know

JonZso, 2020-04-01:
@dr-s-mcgraw ,
Bit late but i managed to try what you suggested and it worked!
Thank you very much

BullzLabz, 2021-07-24:
Hello @JonZso 

I've same issue but with my 3.5” RPi Display (MPI3501).
I'm using Pi B+ (Rev 1.2)

I can't boot to cli.

I don't have `fbcp &` in the `rc.local` file and no result with `CTRL+ALT+F2` (F3, F4, F5...)

Could you help me please ?

skylarweaver, 2021-12-25:
Same issue here! With Ubuntu server 20.04. Nothing I do allows the screen to boot and/or show cli. Starts with white screen, then goes to black. Can ssh in fine.

skylarweaver, 2021-12-25:
Thanks to #305 I have been able to get the terminal to work on ubuntu Mate and have enabled booting directly to terminal with these suggestions: https://ubuntu-mate.community/t/solved-boot-raspberry-pi-2-image-directly-to-terminal-console/1569

But still am unable to do so on ubuntu server 20.04 — please help.

chacalgbi, 2022-08-12:
Same problem. I can't get it to work on Ubuntu 20.04.4 LTS (command line only). Did anyone manage to solve it?