goodtft/LCD-show #182: crashes and kills latest Kavli Linux

SCBuergel, 2020-02-01:
I was running the latest Kavli Linux (I just completed a `apt update` and `apt upgrade -y`) on a brand new Raspberry Pi 4B. My Raspberry is connected to the wifi and I SSH from my laptop into it to install the display. The display called `for Raspberry Pi 4 Touch Screen with Case, 3.5 inch Touchscreen with Fan, 320x480 Monitor TFT LCD Game Display` is already connected in the right way and then followed the instructions posted on Amazon. As you can see in the output below, the install seems to fail, my system shut down and could never be booted again - just the colorful default screen, also loading LEDs of my Rapsberry not saying anything (just red):
```
# git clone https://github.com/goodtft/LCD-show.git
# chmod -R 755 LCD-show/
# cd LCD-show/
# sudo ./MHS35-show
# sudo ./MHS35-show
cp: cannot stat '/etc/rc.local': No such file or directory
(Reading database ... 241937 files and directories currently installed.)
Removing xserver-xorg-input-evdev (1:2.10.6-1) ...
Processing triggers for man-db (2.9.0-2) ...
Reading package lists... Done
Building dependency tree
Reading state information... Done
git is already the newest version (1:2.24.1-1).
The following additional packages will be installed:
  cmake-data librhash0 libuv1
Suggested packages:
  cmake-doc ninja-build
The following NEW packages will be installed:
  cmake cmake-data librhash0 libuv1
0 upgraded, 4 newly installed, 0 to remove and 1 not upgraded.
Need to get 4,857 kB of archives.
After this operation, 24.8 MB of additional disk space will be used.
Get:1 http://kali.download/kali kali-rolling/main arm64 cmake-data all 3.15.4-1 [1,580 kB]
Get:2 http://kali.download/kali kali-rolling/main arm64 librhash0 arm64 1.3.9-1 [121 kB]
Get:3 http://kali.download/kali kali-rolling/main arm64 libuv1 arm64 1.33.1-2 [120 kB]
Get:4 http://kali.download/kali kali-rolling/main arm64 cmake arm64 3.15.4-1 [3,036 kB]
Fetched 4,857 kB in 3s (1,859 kB/s)
Selecting previously unselected package cmake-data.
(Reading database ... 241928 files and directories currently installed.)
Preparing to unpack .../cmake-data_3.15.4-1_all.deb ...
Unpacking cmake-data (3.15.4-1) ...
Selecting previously unselected package librhash0:arm64.
Preparing to unpack .../librhash0_1.3.9-1_arm64.deb ...
Unpacking librhash0:arm64 (1.3.9-1) ...
Selecting previously unselected package libuv1:arm64.
Preparing to unpack .../libuv1_1.33.1-2_arm64.deb ...
Unpacking libuv1:arm64 (1.33.1-2) ...
Selecting previously unselected package cmake.
Preparing to unpack .../cmake_3.15.4-1_arm64.deb ...
Unpacking cmake (3.15.4-1) ...
Setting up libuv1:arm64 (1.33.1-2) ...
Setting up librhash0:arm64 (1.3.9-1) ...
Setting up cmake-data (3.15.4-1) ...
Setting up cmake (3.15.4-1) ...
Processing triggers for libc-bin (2.29-9) ...
Processing triggers for man-db (2.9.0-2) ...
Processing triggers for kali-menu (2020.1.7) ...

Cloning into 'rpi-fbcp'...
remote: Enumerating objects: 61, done.
remote: Total 61 (delta 0), reused 0 (delta 0), pack-reused 61
Unpacking objects: 100% (61/61), done.
CMake Warning (dev) in CMakeLists.txt:
  No project() command is present.  The top-level CMakeLists.txt file must
  contain a literal, direct call to the project() command.  Add a line of
  code such as

    project(ProjectName)

  near the top of the file, but after cmake_minimum_required().

  CMake is pretending there is a "project(Project)" command on the first
  line.
This warning is for project developers.  Use -Wno-dev to suppress it.

-- The C compiler identification is GNU 9.2.1
-- The CXX compiler identification is GNU 9.2.1
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
-- Build files have been written to: /root/LCD-show/rpi-fbcp/build
Scanning dependencies of target fbcp
[ 50%] Building C object CMakeFiles/fbcp.dir/main.c.o
/root/LCD-show/rpi-fbcp/main.c:9:10: fatal error: bcm_host.h: No such file or directory
    9 | #include <bcm_host.h>
      |          ^~~~~~~~~~~~
compilation terminated.
make[2]: *** [CMakeFiles/fbcp.dir/build.make:63: CMakeFiles/fbcp.dir/main.c.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/fbcp.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
install: cannot stat 'fbcp': No such file or directory
2020
need to update touch configuration
dpkg: error processing archive ./xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb (--install):
 package architecture (armhf) does not match system (arm64)
Errors were encountered while processing:
 ./xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb
cp: cannot stat '/usr/share/X11/xorg.conf.d/10-evdev.conf': No such file or directory
reboot now```

goodtft, 2020-05-30:
Do you use Kali linux? the https://github.com/goodtft/LCD-show.git is only for Raspbain sytem. 
For kali Linux driver, see follow:
https://github.com/lcdwiki/LCD-show-kali

kaizoku0506, 2021-07-15:
I tried using the new repo per your instructions, but get the same error.

```
dpkg: error processing archive ./xserver-xorg-input-evdev_1%3a2.10.6-2_armhf.deb (--install):
 package architecture (armhf) does not match system (arm64)
Errors were encountered while processing:
 ./xserver-xorg-input-evdev_1%3a2.10.6-2_armhf.deb
cp: cannot stat '/usr/share/X11/xorg.conf.d/10-evdev.conf': No such file or directory
reboot now
```