goodtft/LCD-show #183: Raspbian Buster Lite (Driver installation without xserver and its dependencies)

KAPMA, 2020-02-02:
Hello!
I am using the latest version of Raspbian Lite.
I encountered the problem that when installing the driver after a reboot, it requires the installation of xserver.
**But at the same time, the display works.**
And I am satisfied with the output of only the command line to the screen.
I do not want to install xserver and its dependencies.
Thanks!

Here is the output that I get during installation:

```
root@raspberrypi:~/LCD-show-master# ./LCD35-show
cp: cannot create regular file '/usr/share/X11/xorg.conf.d/99-fbturbo.conf': No such file or directory
cp: cannot create regular file '/usr/share/X11/xorg.conf.d/99-fbturbo.conf': No such file or directory
2020
need to update touch configuration
Selecting previously unselected package xserver-xorg-input-evdev.
(Reading database ... 39860 files and directories currently installed.)
Preparing to unpack .../xserver-xorg-input-evdev_1%3a2.10.6-1+b1_armhf.deb ...
Unpacking xserver-xorg-input-evdev (1:2.10.6-1+b1) ...
Processing triggers for man-db (2.8.5-2) ...
dpkg: dependency problems prevent configuration of xserver-xorg-input-evdev:
 xserver-xorg-input-evdev depends on libevdev2 (>= 1.2.2+dfsg-1~); however:
  Package libevdev2 is not installed.
 xserver-xorg-input-evdev depends on libmtdev1 (>= 1.1.0); however:
  Package libmtdev1 is not installed.
 xserver-xorg-input-evdev depends on xorg-input-abi-24; however:
  Package xorg-input-abi-24 is not installed.
 xserver-xorg-input-evdev depends on xserver-xorg-core (>= 2:1.18.99.901); however:
  Package xserver-xorg-core is not installed.

dpkg: error processing package xserver-xorg-input-evdev (--install):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 xserver-xorg-input-evdev
reboot now

###

root@raspberrypi:~# apt-get update
Hit:1 http://archive.raspberrypi.org/debian buster InRelease
Hit:2 http://raspbian.raspberrypi.org/raspbian buster InRelease
Reading package lists... Done
root@raspberrypi:~# apt-get upgrade
Reading package lists... Done
Building dependency tree
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 xserver-xorg-input-evdev : Depends: libevdev2 (>= 1.2.2+dfsg-1~) but it is not installed
                            Depends: libmtdev1 (>= 1.1.0) but it is not installed
                            Depends: xorg-input-abi-24
                            Depends: xserver-xorg-core (>= 2:1.18.99.901) but it is not installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

trisax, 2020-02-08:
Hello World
Neither do I want to install Xserver
Help please :-)

MrRikkie, 2020-04-27:
Agree, it works but I do not want to install Xserver, can I uninstall something to remove this error messages ?

goodtft, 2020-05-30:
This Screen driver is suitable for desktop system. The driver can be modified to remove the content related to the desktop system.