goodtft/LCD-show #186: Raspbian buster 2020-02-05: LCD35-show script works with RPi3b, RPi3b+, but not RPi4

thepiam, 2020-02-13:
Hardware
3.5inch TFT SPI screen
http://www.lcdwiki.com/3.5inch_RPi_Display

Tried both the official Rasbpian buster image (2020-02-05) and the image provided on the lcdwiki page with same results.

LCD35-show script works with RPi3b, RPi3b+. Switching back and forth between TFT and HDMI seems to work fine.

For RPi4, LCD35-show works the first time, but after switching back to HDMI display by calling LCD-hdmi script the RPi4 will no longer boot - stuck at rainbow screen.

Pluto8195, 2020-02-15:
Experiencing exact issue described for RPi4, except it doesnt work after reboot even if i dont plug in HDMI display are there any known solutions?

thepiam, 2020-02-15:
It seems like all of these TFTs have very poor software/driver support.
My TFT is this one: http://www.lcdwiki.com/3.5inch_RPi_Display

I assume that it is probably a knock-off of the waveshare 3.5 inch TFT
https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)

However looking at the waveshare repo it seems that even driver support there is pretty bad also:
https://github.com/waveshare/LCD-show/issues

I will probably stay away from these TFTs next time, maybe just get the Adafruit one.