goodtft/LCD-show #187: Uninstall?

rudyauction509, 2020-02-15:
Trying to uninstall, I can’t get “screen configuration” to show up. Stuck with one HDMI port useless and the other stuck at 800x400 resolution. Have way too much stuff on the Pi to wipe it clean. Been fighting with this for at least 8 hours and counting.

lyneca, 2020-03-05:
If the code hasn't changed since I last used it, there is no uninstall. It overwrites system files (/boot/config etc) without making any backups.

xenDE, 2020-03-05:
best way: make clean install on second sd card and copy the not backuped files from there.

goodtft, 2020-05-30:
The uninstall operation will only restore to the state before the installation of the driver, and will not restore to the original state of the system.