goodtft/LCD-show #189: Display Orientation does not match Touch Orientation

KDMcMullan, 2020-02-17:
Display works fine (landscape mode) with no rotation applied. However, the touch sensor appears to operate in "portrait" orientation.

That is to say when I move my finger left/right, the mouse pointer goes up/down and when I move my finger up/down, the mouse pointer goes left/right.

How do I rotate the touch sensor?

DonRichards, 2020-04-02:
Yep, got the same issue.

nikolvi, 2020-04-10:
The same issue here

KDMcMullan, 2020-04-10:
Hello, software people! Your still selling this hardware and folk are still wondering why they gave you their money.

DonRichards, 2020-04-14:
Don't get me wrong, it shows what I need to see. But this is a simple enough correction, I'm not sure why it's either so difficult to implement or it's simply being overlooked.

goodtft, 2020-05-30:
After installing the driver, the display direction and touch direction are the same. For orientation settings, see rotate.sh script

tengfone, 2020-07-18:
got the same issue too, after doing a clean installation still faced with the same problem, tried rotating 0 180 still having same results