goodtft/LCD-show #190: How to use display in console mode only.

pleasemarkdarkly, 2020-02-18:
Installed using the prebuilt images but when I boot into console autologin mode and auto-start htop or a simple program from rc.local or init.d the console output doesn't update. I've tried disabling the rc.local command fbcp& and putting the application before it, but this doesn't work. How do you run console only program with this LCD?

MrRikkie, 2020-04-25:
Did you have any update on this ? I also want to use the display as console

MrRikkie, 2020-04-27:
> Did you have any update on this ? I also want to use the display as console

If you have a RPI image without GUI and install the drivers you get some error messages this it cannot find the directories of the GUI. However t reboot I get the console screen on the display. so no problem here.

goodtft, 2020-05-30:
If you don't need the desktop, you can install the Raspbain Lite system.

MrRikkie, 2020-05-30:
I’m using the lite version now, problem was that if you boot from usb this doesn’t work any more after installation of the drivers. 
The fix is in one of the other posts.

t3kg33k, 2020-11-07:
I am also looking for a console only (without DE) driver for the XPT2048 controller 4.0 display under Ubuntu 20.04 server.
How do we get this driver?

BullzLabz, 2021-07-24:
Same issue, i just need Raspbian Lite version without GUI and some trouble for drivers install.
And i tried with Raspbian GUI, drivers and screen work fine but i can't boot with this screen in console mode !

Please need this screen work with console mode boot 

+1