goodtft/LCD-show #191: LCD35

albfan, 2020-02-19:
Hi, I followed the steps in LCD35-show but there're some mistakes here:

Using sudo in some commands make it to not be able to echo, cp. I can provide a PR for that.

Some dependencies are missed, but maybe is just me not having the desktop installed. we can do some checks to fix that before running it.

Overall the script is really useful. Let me know if you're interested on these contributions.

enricocirignaco, 2020-03-29:
Hi @albfan ,how did you managed to install the drivers? I'm running Raspbian 10 and I tried the following things, without success:
- With Lite version of Raspbian, Desktop Version and Desktop_full version.
- From ssh and via VNC
- run script as root and as pi (if I run it as pi, VNC will be disabled--> "Cannot currently show the desktop" appear.
Maybe you can help me :)

albfan, 2020-03-29:
@enricocirignaco, avoid add noise to a random issue to fix your problem. You should open your own issue for that.

A fex tips:

- if you tried all raspbian OS flavours, then that is not  info for your case, simply Raspbian fails for you
- running from ssh or VNC   is unrelated to make an LCD work with raspberry
- script use sudo, so you should run as normal user.
- Provide exact logs after running the script is how you can get support. Avoid copy/paste what you think is the relevant part for your problem, that only omits info for help you.

Closing this issue as it didn't get progress. Fork is always a good idea if someone is interested.

closed 2020-03-29 by albfan