goodtft/LCD-show #198: Boot on hard drive no longer goes after installing the LCD35-show driver

BelSD, 2020-04-12:
When I put Raspbian Buster Desktop on an SD card and install the LCD35-show drivers, everything works fine.
But when I install Raspbian Buster Desktop on a hard drive and as soon as I install the LCD35-show drivers, as soon as the raspberry restarts, it crashes and there is no way to do anything ...
So there is a bug for me when you want to run your Raspberry on hard drive with a screen that requires drivers ...
I even did the tests with a screen that needs the MPI3508 drivers and it gave the same result -> Blocking at startup ...
[image]
[image]

ErwanBarbier, 2020-04-20:
I have the same issue when booting on USB key

MrRikkie, 2020-04-27:
I also had this issue, took me a day to figure this one out. Would be nice if the installation script takes into account from what device you're booting.