goodtft/LCD-show #199: Lost X on raspi Debian Buster

mgtavares, 2020-04-20:
Solved:

sudo apt-get install xserver-xorg-video-fbturbo

Tks

awhatford, 2020-05-19:
thanks that worked for getting x window up and running but even when using
sudo FRAMEBUFFER=/dev/fb1 startx
the window is still formed on the hdmi output so i guess there is more broken files some ware

awhatford, 2020-05-19:
solved the fb0 only output, edit the option in this file to fb1

/usr/share/X11/xorg.conf.d/99-fbturbo.conf

thanks to this thread
https://forum.repetier.com/discussion/6035/setting-up-joy-it-3-5-display

gabolander, 2021-01-24:
I have a LCD35 for RPi4 connected to a RPi3, and after running LCD35-show, I had the same issue.
I solved in this way:

```
sudo apt-get install xserver-xorg-video-fbturbo
# now I symlink fbturbo driver where it wants to be found ...
sudo ln -s /usr/lib/arm-linux-gnueabihf/xorg/modules/drivers/fbturbo_drv.so /usr/lib/xorg/modules/drivers/
reboot

```
Now X on 3.5" LCD works like a charm ..
On the other hand, I'd like to be suggested to have a virtual keyboard (keyboard on display) so that I can even work totally keybordless. I'm currently working with a usb keyboard and with the touchscreen pen as mouse. 
Tnx in adv. Gabo

SyntaxAerror, 2022-08-14:
@awhatford , I've tried as suggested editing the file and using the FRAMEBUFFER thing, but still get `symbol lookup error: /usr/lib/xorg/modules/drivers/fbturbo_drv.so: undefined symbol: shadowUpdatePackedWeakxinit: giving up` when I try to startx. Am I missing something?