goodtft/LCD-show #200: Turn off or reduce backlight (MPI3501)

Marcaru, 2020-04-25:
Hello

Is there any way to turn off or reduce the brightness of the lcd screen by software?

I have the MPI3501 model.

Bye

MrRikkie, 2020-04-27:
Agree, that would be a nice feature.

DrFredPhD, 2020-05-10:
Also would like this feature

goodtft, 2020-05-30:
Sorry, it needs hardware support to realize this function by software, but there is no backlight brightness control circuit in MPI3501 at present.

MrRikkie, 2020-05-30:
Ok to bad, thanx for your answer.

nouknouk, 2020-06-01:
inhad the same question. thx for the answer.

if I add some control to the power of the display and switch it off (while keeping my rpi zero running), will it resume nicely when i will bring power back to the screen ?

rahmanshaber, 2020-07-03:
I tried out that, if i disconnect all the pins then reconnect then it does not work again, just shows the white display.
but i think disconnecting the power pin will be enough to this trick, but i am not sure if driver will work or not.

rahmanshaber, 2020-07-03:
okay i tested it and i tried disconnecting and reconnecting the power line and it works.
https://youtu.be/0jL5CY34qXk

MrRikkie, 2020-07-03:
Great, i will add a switch in the powerline.

rahmanshaber, 2020-07-03:
> 
> 
> Great, i will add a switch in the powerline.

you have 3.5 inch or others? my one in 4 inch chines.

andreas-globi, 2020-09-25:
If you're powering the display from the Pi, you can switch off USB power on the Pi using https://github.com/mvp/uhubctl

shanness, 2020-09-27:
Thanks @andreas-globi , does that turn of the whole PI or just the screen though?

andreas-globi, 2020-09-27:
It will just turn off the screen (uhubctl let's you control power to the Pi's USB ports)

shanness, 2020-09-30:
Hey @andreas-globi , thanks for the reply.  When I run this command, the screen dims a little for a split second but that's all.  I've tried repeat and other options, but can't figure out how to get it to stay off.  Any help would be much appreciated.  Also, is it possible to turn it back on when touched?
`sudo uhubctl -l 1 -a 0`

S-A-Mayer, 2020-10-18:
@andreas-globi this thread is about the MPI3501 [(Specs)](http://www.lcdwiki.com/3.5inch_RPi_Display), which has a 24-pin interface directly with the RPi's GPIO header.

Is uhubctl also capable of controlling GPIO pins?

andreas-globi, 2020-10-18:
I don't know. It's not my library. I was just sharing to be helpful as it helped me in my case.