goodtft/LCD-show #203: Raspi 4B LCD 3.5 Rpi white screen

wsr13990, 2020-05-05:
Im using LCD 3.5 Rpi on raspi 4B and already do this step:
```
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show
```
After that it stuck on booting but i still able to connect via ssh. The LCD only show white screen.
I already try using Debian Buster fresh install with no success.

vbenner, 2020-05-06:
Same problem from my side

goodtft, 2020-05-30:
Are you used in Debian Buster system？This screen driver is based on Raspbain system. I don't know the specific difference between Debian system and Raspbain system.

dsp77, 2020-06-13:
When I cat /etc/os-release IT says:

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian

What is the difference between Raspbian and Debian you are referring to. Are there any plans to get the display going for 4B?

freitas666, 2021-11-04:
Someone did it works on Debian ?