goodtft/LCD-show #206: Complete failure of xorg displays

awhatford, 2020-05-18:
Hi
Installed lcd35-show and lost working startx and xinit in latist shell only raspbian lite
Just returns no display error. 
The install of xorg was tested working before install of Lcd35-show with manually installed xorg and and matchbox-window-manager tested and working before install

Changes to cmdline and config txt lose existing settings for "i2c rtc" ect rootfs=partitionid=xxxxxxxx changed to rootfs=xxxxxxxx as well

Display dose show in dev under fb1 and command line moves over to tft in boot.
Obviously with no working x window surport I can not see If the touch panel works

Thanks in advance

awhatford, 2020-05-19:
this advice from another poster
sudo apt-get install xserver-xorg-video-fbturbo
works to get the xorg service running but it will only make an x windows on framebuffer 0 even when using sudo FRAMEBUFFER=/dev/fb1 startx
after some digging and another thread 
https://forum.repetier.com/discussion/6035/setting-up-joy-it-3-5-display

the problem was this quote
Option         "fbdev" "/dev/fb0"

in "/usr/share/X11/xorg.conf.d/99-fbturbo.conf"

set this to fb1 and all good

startx launches in fb1 now the tft display

awhatford, 2020-05-19:
also side note had to setup wifi location and country again due to wiped config and cmdline files as well as rtc clock ect

goodtft, 2020-05-30:
Thinks