goodtft/LCD-show #208: Adjust screen

tinwinaung, 2020-05-21:
How can i move screen (a few pixels) to left as right side of screen is hidden

goodtft, 2020-05-30:
I think you can try the "overscan" Settings in the config.txt

MisutaaUrufu, 2021-02-10:
Hey, sorry to post on an old issue, but I'm having this issue as well, only, the overscan settings in config.txt do not work. I've tried various combinations too.

I've tried combinations of `disable_overscan, overscan_scale, overscan_left, overscan_right, overscan_top, overscan_bottom, hdmi_mode, hdmi_group,` and `hdmi_cvt` in config.txt to no results, including setting custom display settings.

I am using a waveshare (or waveshare-like) 3.2" TFT LCD. Raspbian Lite is version 5.4.83 and reports `5.10.11-v7+ GNU/Linux` with `uname -or`.