goodtft/LCD-show #209: CentOS Pi4 drivers

iDJHost, 2020-05-27:
Anyway to add it so I can use the screen for my Pi4 running CentOS? 

I tried to just "install" the drivers and ended up having to reflash my sd card and start over

goodtft, 2020-05-30:
Yes, you can use it for Pi4 running CentOS. But the screen driver we provide is only for Raspbain system, so if you want to use the screen for CentOS system, you need to remake the screen driver for CentOS.
Thanks！

kagesenshi, 2021-02-10:
@goodtft .. by "remake", do you mean changing the LCDXX-show script to CentOS/Fedora or do you mean building back the dtb file?.

I'm trying to get this to work with Fedora ARM, at least for the MPI3501, and would be great if i can have some pointers.

1 - would the dtb files work with any distro?
2 - any non-upstream kernel module that is needed?, or is the dtb file alone is enough?