goodtft/LCD-show #218: White screen raspberry PI ZERO WH

DanielSergheenco, 2020-07-02:
Hello! Help me update this driver for raspbery pi zero wh.. 
I also tried older commits and LCD still remains white.

[image]
[image]

billyjacoby, 2020-07-19:
Having the same issue with the exact same screen on a pi 3B+, haven't been able to figure it out yet,

miticoEffandi, 2020-09-15:
> Having the same issue with the exact same screen on a pi 3B+, haven't been able to figure it out yet,

hey, did you got it work? i have the same problem. thanks

billyjacoby, 2020-09-15:
Mine actually ended up being an old screen that was bad. Bought and wired up a new one and it worked fine. 

Sorry I couldn’t be more help!