goodtft/LCD-show #219: After driver installed on a Raspberry 3b+ with LCD35-show wont boot

thevinayaknair, 2020-07-06:
After driver installed on a Raspberry 3b+ with LCD35-show wont boot, it is stuck at "Raise Network Interfaces" in case it helps. I'm also using  "Raspberry Pi OS"

Raspberry Pi OS (32-bit) with desktop and recommended software
Image with desktop and recommended software based on Debian Buster
Version:May 2020
Release date:2020-05-27
Kernel version:4.19
Size:2523 MB
[image]

pheiduck, 2020-07-12:
also had this, resolved by installing `xserver-xorg-input-evdev` first. Also did a merge request for updating the readme:
https://github.com/goodtft/LCD-show/pull/221

pheiduck, 2020-07-12:
otherwise the script needs to be recoded

sagd330, 2020-07-14:
I had the same problem. Do you know if it is possible to recover the system, or access in some way to the documents from a PC?

pheiduck, 2020-07-15:
In my experience I had to reimage the SD-Card or USB Device. But on Raspbian OS Lite x32 it worked for my after installing `xserver-xorg-input-evdev` first

elewin, 2020-08-08:
I am having the same problem. I can still SSH in to the device, and if i do ctrl-f2/f3 etc it will go to a new tty, but the default one is stuck. Interestingly, if i do something like type ls, it will blink in the screen for like 1 frame and go back. It's strange.

elewin, 2020-08-08:
I may have found a workaround, open up /etc/rc.local and comment out the line `fbcp &`

OsirianOne, 2020-08-13:
> I may have found a workaround, open up /etc/rc.local and comment out the line `fbcp &`

How without a pc or other Linux to access the card how to do this. 

Reimaging every time something goes wrong with an RPi makes me went to their it in the trash.

Kyllingene, 2020-09-05:
I have the same problem, except xserver-xorg-input-evdev didn't work, ended up stuck just like normal.

elewin, 2020-12-04:
> > I may have found a workaround, open up /etc/rc.local and comment out the line `fbcp &`
> 
> How without a pc or other Linux to access the card how to do this.
> 
> Reimaging every time something goes wrong with an RPi makes me went to their it in the trash.

You can connect to it over SSH from another device, even a smart phone. Just make sure you enable SSH when you do your initial setup

OsirianOne, 2021-02-02:
It took forever to figure out.
The overlay needs referenced in the boot. Txt file before running the
script and the instructions so not include how to do this accurately, nor
does it explain the 0, 1, 2 for the framebuffer to make it primary
secondary or tertiary display.

I figured it out all myself and took months.

Kyllingene, 2021-02-03:
Sorry, I'm a noob when it comes to Linux stuff and command line. How would one reference the overlay  in the boot?

OsirianOne, 2021-02-03:
Just put the name of which overlay that you want loaded in the line,
position does not matter only that is is a space before and after to
separate it from other variables.

Kyllingene, 2021-02-04:
So I edit ```fbcp &``` to be ```fbcp & (0,1,2, or 3)```?

OsirianOne, 2021-02-04:
I do not remember what allowed me to set the fb to make the regular HDMI as
the primary, your script sets the TFT as the primary.  In some cases that
is wanted.  In others it isn',t it depends on the application for the
screen and whether it is touch or just the 3.5.

I plan to eventually either integrate it into a hardshell case for a
Raspberry Pi System (CyberDeck
<https://hackaday.com/2020/03/10/surviving-the-apocalypse-with-a-briefcase-full-of-pi/>).
The touchscreen will provide access and operate internal servos to open or
release clamps as a script, then set the screen to display gauges and tiny
tabs that have other gauges, temp, batt, interface to set the power
conservation, and give output for recharge time or consumption and
remaining time.
But my primary 15.5" 1920x1080, to go in the lid that gets released and
that script before will have a line to turn that on using a GPIO pin to
trip a solenoid or a relay that opens the power to the display (not touch
though).

Then of course use the RPi or the 3.5 screen and set everything to standby
or sleep, etc.

1)Get the LCD-Show git hub project

git clone https://github.com/goodtft/LCD-show.git

cd LCD-show

2) Boot Overlays - Copy over only the 1 file that is needed for this project

Copy over boot overlays

In the original script, the same file is copied over under two different
names. In the next step (“Enable interfaces”), the .dtbo file is
referenced, albeit without the file extension, so only that file should be
required.

The file does not exist in the /boot/overlays directory, so it's safe to
copy it over.

*sudo cp ./usr/tft35a-overlay.dtb /boot/overlays/tft35a.dtbo*

DO NOT COPY OVER YOUR CONFIG.txt FILE nor your commandline.txt -no NO NOOOOOOO-

Enable interfaces

A particular boot config file is copied over. The changes are minimal, and
to avoid making unexpected changes, the changes should be made manually.
# Don't run. See notes below to make changes manually.
# sudo cp ./boot/config-35.txt /boot/config.txt  (THIS IS THE BIGGEST
PROBLEM, files need modified not replaced)

Using raspi-config, enable the following interfaces:

Open a Terminal and type *sudo raspi-config*

   - SPI
   - I²C (may not be needed, see below)
   - Serial (may not be needed, see below)

*Edit:* Reddit user farptr points out that SPI is the one used for driving
the display, and the I²C interface is used for the touchscreen.
<https://www.reddit.com/r/raspberry_pi/comments/abb40e/i_spent_the_holidays_setting_up_my_new_lcd_screen/ed0yj07>
I experimented with the interfaces, and in my case, *only SPI was required
for both the display and touchscreen functionality*. I recommend turning
these interfaces on one by one in to see which ones your display needs.

Uncomment the line that says :
# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

[... 49 lines cut ...]

   - Leave the console=serial0,115200 option as is. The reference file
   changes serial0 to ttyAMA0. The Raspberry PI has two UARTs, and ttyAMA0
   refers to one of them. However, the primary UART differs based on the
   Raspberry Pi model (namely based on if the model comes with built-in
   Bluetooth and WiFi). serial0 can be used to reference the primary one
   for the current model.
      - *Leave the root option as is. The reference file specifies
      /dev/mmcblk0p2 as the root partition, but newer versions of
Raspbian refer
      to the root partition by a UUID. In fact, this is the difference between
      the non-NOOBS and NOOBS version of the reference file, so simply avoiding
      this change means you don’t have to worry about which distribution you’re
      using.*
   - Don’t remove the fsck.repair=yes option. This is unrelated to the LCD,
   which is why we won't bother removing the option.
   - Don’t add the logo.nologo option. Again, this is unrelated to the LCD,
   so if you want to keep the default behavior, there’s no need to remove the
   logo.

Reboot

At this point, the TFT LCD is set up. If you just want a display for the
console, you can reboot.
*sudo reboot *

Kyllingene, 2021-02-05:
That seems like a lot to do just to make a touchscreen work. I really hope that the devs fix this soon... I probably won't for now, since I don't *need* the touchscreen, it would just be nice to have.

OsirianOne, 2021-02-05:
> that the devs fix this soon... I probably won't use it for now, since I

No it's not it is like 4 steps and save s diskspace.

But yes they need to build a interactive script that updated this files and
does the moves for the user and then it does the steps and merges them into
existing system files, instead of replace, but a lot of linux users are
used to these types of step s (when they have the them correct and done
have to guess)

Now the FB is in the x.org server settings, now that's confusing. LOL