goodtft/LCD-show #223: Display remains white after full-upgrade

PureFallen, 2020-07-20:
After installing a full-upgrade as part of a maintenance, the display i used ./LCD35-Show for remains white. Reinstalling the driver wasn't helpful, but i know the display is likely to be healthy, there is probably just driver issues.
Any suggestions how i could tackle the issue? Is there a way to manually remove the driver and give it another shot?
Are there any dependancies the full-upgrade broke? (Can't really imagine that, `xserver-xorg-input-evdev` is inside the repo after all)

pheiduck, 2020-07-20:
Can confirm, same on my end. Reinstall doesn’t help, happend because of the 5.4 kernel so the LCD Driver needs to rebuild?

pheiduck, 2020-07-20:
I think the Kernel update broke this :/

pheiduck, 2020-07-20:
Downgrade to 4.19 resolved this for me:
rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2

PureFallen, 2020-07-20:
> 
> 
> Downgrade to 4.19 resolved this for me (temporarily):
> `rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2`

Thanks for the information!
The display has low priority for me, displaying the status of my pi-hole but i have in worst case SSH and the website for that.
Guess we could try to work on rebuilding the LCD Driver? But yeah, @goodtft needs to know about that issue.

pheiduck, 2020-07-20:
> displaying the status of my pi-hole

same to me

jjaaccoobb, 2020-07-23:
> Downgrade to 4.19 resolved this for me (temporarily):
> `rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2`

Cheers! This is the only thing that works for now.

PureFallen, 2020-07-23:
Downgraded to 4.19 using `rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2` today for testing purposes.
Display still not working for me, i probably broke something else in the meantime where i tried to fix the issue.

pheiduck, 2020-07-23:
@PureFallen Which model do you have? - I have Raspberry Pi 3 B+ works without issues 
I recommend `rpi-update 2d76ecb08cbf7a4656ac102df32a5fe448c930b1 `

pheiduck, 2020-07-23:
>i probably broke something else in the meantime where i tried to fix the issue.

What do you tried?

jaymil3, 2020-07-23:
> @PureFallen Which model do you have? - I have Raspberry Pi 3 B+ works without issues
> I recommend `rpi-update 2d76ecb08cbf7a4656ac102df32a5fe448c930b1 ` for Raspberry Pi 4 User (and 3)

Hi - i have a pi 4 with 3.5" touchscreen, fresh install of OS and tried both rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2 and 2d76ecb08cbf7a4656ac102df32a5fe448c930b1.  the first one seems to finally work as a display with ./MSH35-show, but the touchscreen will not respond.  i've tried several other options like ./MPI35-show, etc. and those don't work at all (white screen).  this is really frustrating - any thoughts?

pheiduck, 2020-07-24:
@jaymil3 try ./LCD35-show should work then with 4.19

jaymil3, 2020-07-24:
> @jaymil3 try `./LCD35-show` should work then with 4.19

thanks @pheiduck - sigh. apparently one of my screens is defective and touch doesn't work on it, but it does on the other.  i'm successfully using rpi-update 2d76ecb08cbf7a4656ac102df32a5fe448c930b1 with ./MHS35-show now.

PureFallen, 2020-07-24:
> 
> 
> @PureFallen Which model do you have? - I have Raspberry Pi 3 B+ works without issues
> I recommend `rpi-update 2d76ecb08cbf7a4656ac102df32a5fe448c930b1 ` for Raspberry Pi 4 User (and 3)

Raspberry Pi 3B+ for me.
Tried to reset with LCD-hdmi before and clones LCD-Show from the repo here again for potential updates.
Guess something didn't work there as expected.

rogerlucas-veea, 2020-07-25:
Confirmed for me as well with a 3B+.  White screen on the latest 5.4.5l-v7+ kernel, working with e1050e94821a70b2e4c72b318d6c6c968552e9a2 on the 4.19.118-v7+ kernel.

Anything we can do to help diagnose the issue?

pheiduck, 2020-07-25:
> Anything we can do to help diagnose the issue?

I hope so, some developers can do and at a PR or fork this with patches for the 5.4 Kernel

PureFallen, 2020-07-26:
`rpi-update 2d76ecb08cbf7a4656ac102df32a5fe448c930b1` worked for me while `rpi-update e1050e94821a70b2e4c72b318d6c6c968552e9a2` did not, raspberry Pi 3B+.
Could be a different revision that is causing that, but whatever reason is has: Kinda tells me again why i don't like to mess around with rpi-update.
For the sake of testing that is probably good to know, yet i hope in the long runt hat the Display Driver gets fixed and we no longer need to rely on inconsistent downgrades of the Kernel.

aberfisch, 2020-07-26:
+1 Raspberry Pi 2 Model B Rev 1.1

MrYacha, 2020-07-26:
I spent a few hours and fixed it. Take a look into https://github.com/MrYacha/LCD-show.

micware, 2020-07-27:
Raspberry Pi 2 Model B now works fine, thank you :D

rogerlucas-veea, 2020-07-27:
Hi @MrYacha, thanks for the above fix.  It works on the older e1050e94821a70b2e4c72b318d6c6c968552e9a2 and on the newer 5.4.51-v7+ as installed by `rpi-update` without any options.

I noticed that the screen refresh is much slower with the new driver.  With the original LCD35-show config and the older kernel, the screen refresh was very fast - good enough to watch video - but with the updated driver, the refresh is much slower and you can actually see the screen updating.  I checked the DTB changes and the clock rate has been reduced from a maximum of 115 MHz down to 16 MHz, so the screen refresh is about 7x slower.

I've tried to investigate this and try with faster SPI clock rates but the DTC operation fails as below.

```
pi@octopi:~ $ dtc -O dtb -o new-115MHz.dto new-115MHz.dts
new-115MHz.dto: Warning (unit_address_vs_reg): /fragment@0/__overlay__/spidev@0: node has a unit name, but no reg property
new-115MHz.dto: Warning (unit_address_vs_reg): /fragment@0/__overlay__/spidev@1: node has a unit name, but no reg property
new-115MHz.dto: Warning (unit_address_vs_reg): /__local_fixups__/fragment@2/__overlay__/waveshare35a@0: node has a unit name, but no reg property
new-115MHz.dto: Warning (gpios_property): /fragment@2/__overlay__/waveshare35a@0:reset-gpios: cell 1 is not a phandle reference
new-115MHz.dto: Warning (gpios_property): /fragment@2/__overlay__/waveshare35a@0:reset-gpios: Could not get phandle node for (cell 1)
new-115MHz.dto: Warning (gpios_property): /fragment@2/__overlay__/waveshare35a@0:dc-gpios: cell 1 is not a phandle reference
new-115MHz.dto: Warning (gpios_property): /fragment@2/__overlay__/waveshare35a@0:dc-gpios: Could not get phandle node for (cell 1)
new-115MHz.dto: Warning (gpios_property): /fragment@2/__overlay__/waveshare35a-ts@1:pendown-gpio: cell 1 is not a phandle reference
new-115MHz.dto: Warning (gpios_property): /fragment@2/__overlay__/waveshare35a-ts@1:pendown-gpio: Could not get phandle node for (cell 1)
new-115MHz.dto: Warning (gpios_property): /__fixups__:gpio: property size (242) is invalid, expected multiple of 4
dtc: livetree.c:565: get_node_by_phandle: Assertion `generate_fixups' failed.
Aborted
pi@octopi:~ $
```

How did you compile the DTS to DTB and was the reduction in clock rate a deliberate change to get the display to work?

Cujo31ch, 2020-07-27:
I have done what is mentioned on step3 here on my Pi 4 but it didn't helped still white display.
http://www.lcdwiki.com/3.5inch_RPi_Display
I have a display from Joy-It with this controller
ControllerXPT2046

rogerlucas-veea, 2020-07-27:
Hi @MrYacha, I discovered that the issue with compiling the changes is that the version of "dtc" provided with Rasbian is too old.  I downloaded the latest release of DTC (https://git.kernel.org/pub/scm/utils/dtc/dtc.git/tag/?h=v1.6.0), compiled it and that worked fine.

I have updated my config to restore the 115 MHz speed ("spi-max-frequency = <0x06dac2c0>;") and it's still working for me.  

If anyone wants to test the changes, use the above link from @MrYacha to his forked repo, download it, run the `MHS35-show` installer then update your `/boot/overlays/waveshare35a.dtbo` with the one in ZIP archive attached to this comment.

[waveshare35a-115MHz.zip](https://github.com/goodtft/LCD-show/files/4981984/waveshare35a-115MHz.zip)

Cujo31ch, 2020-07-27:
I have now created a new micro SD with the image at the bottom of this page and then it is working.
http://www.lcdwiki.com/3.5inch_RPi_Display
The question is now what is not okay with my image as I have 4 applications running which are working and I wouldn't do a new installation if possible.

Cujo31ch, 2020-07-27:
Does anybody has an idea how I can find out what is not good with my installation after doing step 3 of the wiki page?

MrYacha, 2020-07-27:
Hi @rogerlucas-veea
>  and it's still working for me
Unfortunately, my screen stays white with your dtbo.

rogerlucas-veea, 2020-07-27:
@MrYacha thanks for the feedback... that's very interesting because it's the original maximum clock rate from the earlier DTS.  

One of the advantages of these displays is that they can run at higher frame rates to support video playback (compared to the I2C ones that are limited to a few frames per second).  We lose that if we have to drop the clock down to 16 MHz.

Could you try some different clock rates - perhaps 80 / 60 /40 / 20 MHz to see how high you can set your clock without the display failing and post the results and the DTO files here?

That would allow others to select the fastest clock rate that their hardware can use and get the best frame rates.

MrYacha, 2020-07-27:
@jjaaccoobb I have made few versions (20, 60, 80), 20 and 60 are working perfectly while 80 is not. I made a archive with prebuilt dto files and instructions/base dts file to compile new ones.
https://ufile.io/vaeet36o
for those guys which don't want spend time on this I still have a forked repo with fixed dto - https://github.com/MrYacha/LCD-show

pheiduck, 2020-07-28:
@MrYacha your fork works for me :)

Cujo31ch, 2020-07-28:
I checked again the Image solution from the wiki to see why it is not working with my updatet OS.
And I see that the solutions on this wiki are with Kernel 4.19 and I have an updatet Kernel 5.4.
Is it correct that there is still no solution for Kernel 5.4?

rogerlucas-veea, 2020-07-28:
Hi @Cujo31ch , there is a fix done by @MrYacha above that works with the latest 5.4 kernels.  Please use https://github.com/MrYacha/LCD-show rather than the this LDC-show and install in the usual way.  Your display should now work but the frame rate will be a bit lower than before.  

If you want to try different clock rates to see how fast your display will go, @MrYacha has also done a set of configurations at different clock rates for you to test.  Download the TAR archive here: https://ufile.io/vaeet36o to your Pi.  Then untar it and try the different dtbo/dto files to see which is best for you. 
e.g.
```
tar -xf lcd.tar
sudo cp 115.dtbo /boot/overlays/waveshare35a.dtbo
sudo reboot

```
If this doesn't work, try the next speed down...

```
sudo cp 80.dto /boot/overlays/waveshare35a.dtbo
sudo reboot
```
etc.

The LCD35 is a generic design made by lots of different manufacturers and some work faster than others.  My display seems to run OK at the 115 MHz, the one belonging to @MrYacha doesn't work at 80 MHz but is fine at 60 MHz.

The default setting is for 15 MHz, which is slow but safe - you should have no problems with this speed, but the display refresh isn't very fast.  If you can run the display at a faster clock rate, the refresh rate will improve.

If you find that this works and you try the above speed tests, please can you report back how fast your display can go?  It would be good to have feedback on the display performances so that the default configuration can be optimised...

Good luck.

Cujo31ch, 2020-07-28:
As far as I understood I need to do what is mentioned under step 3 for my Joy-IT Display.
http://www.lcdwiki.com/3.5inch_RPi_Display
But this doesn't fix my issue. The Display still shows a white screen.

rogerlucas-veea, 2020-07-28:
@Cujo31ch Please log into your Pi and do the following:

```
sudo rm -rf LCD-show
git clone https://github.com/MrYacha/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show
```

Note that the HTTPS URL is different.  @MrYacha has a fix, but it is not in this repo yet, it is still only on his fork.

The above should fix your issue.

Cujo31ch, 2020-07-28:
Okay now I got it and it worked. The Display is working again. Thanks for the help

o5, 2020-07-28:
I'm a RPi newbie (LCD arrived two weeks ago from ALI and RPi 4B 2GB one week ago). In previous weekend I played with RPi and discovered how to use it (prepare SD and remote connect) and ended with white screen. It's really funny, because it looked very easy in many articles and videos but unfortunately new kernel has been released, so I tried it in bad time. Luckily I found this thread and hint about downgrade to kernel 4.19 and it finally started to work!

I'm currently on kernel `5.4.51-v7l+` and @MrYacha version of LCD-show. Tried 115MHz, 80MHz but didn't work. 60MHz works. I've [this](https://www.aliexpress.com/item/32834124804.html) LCD.

xevilstar, 2020-07-29:
downgrade is not a solution is a workaround the drive should work with the
current environment

kesar, 2020-07-29:
> I spent a few hours and fixed it. Take a look into https://github.com/MrYacha/LCD-show.

Could you do the same for 3.2? Thanks!

bfagundez, 2020-07-30:
I have a MHS35 display and is not working with tne new kernel either, I've tried @MrYacha fork but no luck.
Only reverting like @pheiduck suggests work. 
I see other issues:
I just want to run in text mode but the screen does not seem to clear the characters after the boot sequence, is that expected?

twerthi, 2020-07-31:
> I have a MHS35 display and is not working with tne new kernel either, I've tried @MrYacha fork but no luck.
> Only reverting like @pheiduck suggests work.
> I see other issues:
> I just want to run in [...]

Same here, @MrYacha fork didn't work for me either.

rogerlucas-veea, 2020-07-31:
Hi @bfagundez and @twerthi , can you tell us what version of Raspberry Pi you are running and where you sourced your MHS35 displays from?  We know that some combinations of Pi + display do work... with varying performance... so if there are some that don't work, it would be good to have as much information as possible so that we can help track down the issue that you are experiencing.
Thanks!

twerthi, 2020-07-31:
I'm using the Raspberry Pi 4, 4 GB model.  I have purchased two of the displays, both from Amazon.  The display had been working fine up until this week.

Cujo31ch, 2020-07-31:
Have you done this?
sudo rm -rf LCD-show
git clone https://github.com/MrYacha/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show

Perhaps only sudo ./LCD35-show is different as you have another display. My display is from joy-it but I also have a Pi 4 and this fixed my issue.

twerthi, 2020-07-31:
These are the directions from the Amazon product
sudo rm -rf LCD-show
git clone https://github.com/goodtft/LCD-show.git
chmod -R 755 LCD-show
cd LCD-show/
sudo ./MHS35-show

I put these instructions in a bash script, commented clone for goodtft and added a new line with the one from MrYacha.  It rebooted, but the screen remains white.

MrYacha, 2020-07-31:
> These are the directions from the Amazon product
> sudo rm -rf LCD-show
> git clone https://github.com/goodtft/LCD-show.git
> chmod -R 755 LCD-show
> cd LCD-show/
> sudo ./MHS35-show
>
> I put these instructions [...]

@twerthi Can you try to remove mention of old dts in the /boot/config.txt?

twerthi, 2020-07-31:
@MrYacha, forgive me, my *nix knowledge is still pretty low.  This is what my config.txt looks like (there are a bunch of commented out lines above this, I didn't think they were relevant so I didn't include them).  What is it I need to remove?

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

[pi4]
# Enable DRM VC4 V3D driver on top of the dispmanx display stack
#dtoverlay=vc4-fkms-v3d
#max_framebuffers=2

[all]
#dtoverlay=vc4-fkms-v3d

# NOOBS Auto-generated Settings:
hdmi_force_hotplug=1
dtparam=i2c_arm=on
dtparam=spi=on
enable_uart=1
dtoverlay=mhs35:rotate=90
hdmi_group=2
hdmi_mode=1
hdmi_mode=87
hdmi_cvt 480 320 60 6 0 0 0
hdmi_drive=2

kiokoman, 2020-08-01:
@MrYacha fork worked for me,  but /usr/mhs35-overlay.dtb is missing from his fork, I had to rename/copy waveshare35a.dtbo to mhs35-overlay.dtb and run sudo ./MHS35-show, after rebooting everything was ok

MrYacha, 2020-08-01:
Hi @twerthi 
Instead of:
> dtoverlay=mhs35:rotate=90
You should write this:
`dtoverlay=waveshare35a:rotate=90`

twerthi, 2020-08-01:
Eureka!  @kiokoman your solution worked for me, @MrYacha thank you for making the fork 👍

jaymil3, 2020-08-02:
hi - confirming that the MrYacha fix for MHS35 works on pi4 with full upgrade (5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux).  the screen i'm using is from [Tangyy on Amazon](https://www.amazon.com/gp/product/B07WRV48ZW/ref=ppx_yo_dt_b_asin_title_o03_s00?ie=UTF8&psc=1) 

Thanks!!

however changing the display resolution to something other than the default 720x480 in the raspberry pi config gui has no effect - the display appears exactly the no matter the resolution choice.  prior to this problem i was able to use 640x480, and was hoping to be able to do that with this update.  that resolution is blurry so i was also side-hoping that i could fix that with different clock rate settings that you've been discussing, but i can't even get the resolution setting to change.

thoughts?

elentophanes, 2020-08-03:
Thank you so much @MrYacha. That worked perfectly on an old B+ version 010 fully upgraded. 
Linux raspberrypi 5.4.51+ #1327

I used:

sudo rm -rf LCD-show
git clone (https://github.com/MrYacha/LCD-show.git)
chmod -R 755 LCD-show
cd LCD-show/ 
sudo ./LCD35-show 
It overwrote the bad driver by cloning your fork into the same directory and rebooted to my 3.5” RPi Display (MPI3501). With the touch screen and a virtual key board working.

GitFun4All, 2020-08-07:
Thank you @MrYacha, 60hz works for me using lcd-35 on a Pi4. Cheers!

PureFallen, 2020-08-09:
> Thank you so much @MrYacha. That worked perfectly on an old B+ version 010 fully upgraded.
> Linux raspberrypi 5.4.51+ #1327
>
> I used:
>
> sudo rm -rf LCD-show
> git clone [...]

Worked for me, now i am on 5.4.51 #1327 myself.
It is required to remove old entries in the boot.config. There might be the way to restore the boot.config from "before installing the driver" by running `sudo ./LCD-hdmi` before deleting the old folder, yet this only seems to be a good idea if you did no other changes to the boot.config in the meantime.

Let us see if goodtft will answer to this at all, or if we will rely on MrYacha in the long run. Guess thats the price you pay by saving a few bucks with a China Display that redirects to this driver here in the manual, yikes.

jledgett, 2020-08-10:
MrYacha's fix seemed to correct the white screen issue, but I can't seem to tweak the resolution like I could before.   

This results in no simultaneous output to my monitor.

These settings in the /boot/config.txt seem to have no effect.

```
hdmi_group=2
hdmi_mode=85
hdmi_cvt 1280 720 60 0 0 0
hdmi_drive=2
```

juleswnfld, 2020-08-12:
Hi, for 3.5inch screen in landscape position (90 degrees rotation), if anybody needs right click on stylus long-press, you can add the following lines in ./usr/99-calibration.conf-35-90 (for other screens I guess it should work in corresponding files):

   Option "EmulateThirdButton" "1"
   Option "EmulateThirdButtonTimeout" "750"
   Option "EmulateThirdButtonMoveThreshold" "30"

Tested it on 4B with Buster with 3.5 LCD hat, working ok.

Thanks @MrYacha for his work. had the white screen problem, his branch solved my issue.
Maybe you could incorporate this right-click option in your code. It is pretty useful.

juleswnfld, 2020-08-12:
Also, if you keep your HDMI port connected at all times to a monitor when you actually want video output on the LCD hat (gpio connection), the monitor is on but not displaying anything or remains blocked in boot screen.
To actually turn it off, I found that running : $ vcgencmd display_power 0 
will turn off the monitor, which is desirable in my case.

To turn the power for hdmi back on: vcgencmd display_power 1

PS:
I created some scripts to call LCD-hdmi / LCD35-show and created icons in Main Menu to easily switch between screens, depending on what I use the Pi for...

jledgett, 2020-08-14:
@juleswnfld cool.   Do you have any idea how to use alternate resolutions like you used to be able to?

juleswnfld, 2020-08-14:
> @juleswnfld cool. Do you have any idea how to use alternate resolutions like you used to be able to?

No, haven't played with resolutions, I'll look and report back if I find something.

amanchawla123, 2020-08-17:
Resolved with @MrYacha repo @ 60MHz.

Raspi used - 4B 2GB.
LCD used - Waveshare 3.5in LCD

Steps -

1. Move kernel to latest version 5.4.58-v7l+
2. Forked @MrYacha repo as follows -
**_sudo rm -rf LCD-show
git clone (https://github.com/MrYacha/LCD-show.git)
chmod -R 755 LCD-show
cd LCD-show/
sudo ./LCD35-show_**
3. Verified if boot/config had the following text - **_dtoverlay=waveshare35a:rotate=90_** , if yes then move forward. Otherwise manually add and comment the text - **_**dtoverlay=mhs35:rotate=90**_**
4. Download lcd.tar from https://ufile.io/vaeet36o
Use the following commands -
**_sudo cp 60.dto /boot/overlays/waveshare35a.dtbo
sudo reboot_**

LCD worked. Thanks @MrYacha

m8asolutions, 2020-08-21:
> I spent a few hours and fixed it. Take a look into https://github.com/MrYacha/LCD-show.

MrYacha, thanks so much for this. I spent all day working on this.

bluehash, 2020-09-03:
@MrYacha Could you please reupload your files for the higher speeds? Thank you!

> @rogerlucas-veea I have made few versions (20, 60, 80), 20 and 60 are working perfectly while 80 is not. I made a archive with prebuilt dto files and instructions/base dts file to compile new ones.
> https://ufile.io/vaeet36o
> for those guys which don't want spend time on this I still have a forked repo with fixed dto - https://github.com/MrYacha/LCD-show

bluehash, 2020-09-03:
I'm fortunate to have a couple of different displays and may shed some light on why @rogerlucas-veea's 115MHz works on his. I'm also able to get only upto 20Mhz on MrYacha's fork.

I'm here because the refresh rate on the 5.4kernel is extremely slow and defaults to 16Mhz when you look at dmesg:
[image]

There are many manufacturers. The cheaper ones use the ILI9486 driver. There is a slow one(16Mhz) and a fast one(125Mhz). I have the LCD(A) which is slower and the MHS35, which is the faster one. Pics at the end of the post.

1. The dtbo from the waveshare site(https://github.com/waveshare/LCD-show.git) on the LCD(A)
I can get to a speed of 16MHz.
When speed is increased in the /boot/config.txt above 20MHz, `dtoverlay=waveshare35a:speed=20000000`, colors are off and you see a lot of flickering.
[image]

2. rogerlucas-veea's 115Mhz dtbo works on the MHS35, but not on LCD(A). LCD(A) shows a white screen.
[image]

3. MrYacha's dtbo works on LCD(A). It defaults to 16Mhz, but can be clocked to 20MHz via config.txt. Above 20, the colors get washed out and the screen flickers.

20Mhz
[image]

It also works on the MHS35, with no issues @115MHz
[image]

So, the best thing working for me so far is either the waveshare or MrYacha's dtbo, clocked at 20Mhz... for the LCD(A)
If you have the MHS35, you can use rogerlucas-veea's 115MHZ dtbo or clock up  MrYacha's dtbo

LCD(A)
[image]

MHS35
[image]

Hope this helps.

[13 more replies]