goodtft/LCD-show #234: Display not working, Maker Factory 3.5inch RPi Display

raiserle, 2020-08-19:
I bought this display from CONRAD named as TRU COMPONENTS.
https://www.conrad.de/de/p/tru-components-touchscreen-monitor-8-9-cm-3-5-zoll-320-x-480-pixel-2144982.html
But on the back of the board is written MAKER FACTORY and I think it is the MAKEVMP400.
After installation, nothing is shown on the display. 
I used `LCD35-show`.

- RasPi 3B+
- Raspberry Pi OS Full (32-bit)
- Kernel 4.19 (updated)
- Release Date "2020-05-27
- CodeName: Buster
- Latest updates installed => `uname -r` => 5.4.51-v7+ | `cat /etc/issue` => Raspbian GNU/Linux 10

danielktdoranie, 2020-08-20:
Having the same issue

raiserle, 2020-08-20:
@danielktdoranie can you confirm this?
Without update the Raspbian to the new 5.x Kernel, the display works fine!

Workarround: Don't update the Raspbian! Load the NOOBS 3.4 release and install this without any internet connection to block the auto-update after reboot.

danielktdoranie, 2020-08-20:
> @danielktdoranie can you confirm this?
> Without update the Raspbian to the new 5.x Kernel, the display works fine!
> 
> Workarround: Don't update the Raspbian! Load the NOOBS 3.4 release and install this without any internet connection to block the auto-update after reboot.

That’s a pretty subpar work around LOL.

Yes, I can confirm. I can confirm the same for 32-bit and 64-bit Ubuntu MATE for Raspberry Pi as well as the 64-Bit version of Raspberry Pi OS.

This code needs to be maintained and updated. I got mine from Amazon.co.uk looks like I will sending it back

tux1c, 2020-08-28:
> Yes, I can confirm. I can confirm the same for 32-bit and 64-bit Ubuntu MATE for Raspberry Pi as well as the 64-Bit version of Raspberry Pi OS.

you should check this out for 64bit OSes:
https://github.com/tux1c/wavesharelcd-64bit-rpi

it might work on 32bit too, I'm still unsure if dtbo files are architecture specific or not.

bierbo, 2020-09-01:
> you should check this out for 64bit OSes:
> https://github.com/tux1c/wavesharelcd-64bit-rpi
>
> it might work on 32bit too, I'm still unsure if dtbo files are architecture specific or not.

This does also work for Raspberry Pi Zero WH (32 Bit Raspbian)

cgomesu, 2020-09-03:
there's a pull request (see the one by @TheAlienDrew) that fixed it for me (`LCD35-show` on RPi 3B with Kernell 5.4.51). you might want to try it manually or wait until it's merged.

eat-sleep-code, 2020-09-18:
The repo has been updated about a week ago with a fix for this issue.

amarnath123456789, 2023-10-07:
> This does also work for Raspberry Pi Zero WH (32 Bit Raspbian)

Hi , Im facing the same white screen problem on my 3.5 inch TFT display , I'm using waveshare driver and still nothing happens , Im using the standard raspberry os which was said recommended mostly 32 bit ig. Im a newbie so what should I do exactly?